LinuxCNC/linuxcnc #231: Axis interface freeze on joint limits (joint_axes16)

grandixximo, 2017-02-12:
## Here are the steps I follow to reproduce the issue:

 1.Build and install joint_axes16 branch
 2.Open the Vismach 5Axis mill machine
 3.Home and Jog B and C, then move machine near a limit
 4.Machine stops near a limit, and send message about the limit, and a hint to switch to Joint mode to jog out of the limit

## This is what I expected to happen:

It should let you switch to Joint mode to jog out of the limit

## This is what happened instead:

The Axis interface freeze, and it does not let me switch to joint mode or jog.

## It worked properly before this:

Sorry, but i'm not sure about which version behaved properly

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): 2.8.0-pre1~ja
 * I am using this kernel version (shown by `uname -a`):4.6.7-rt11
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [x] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):joint_axes16
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):Axis
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): none

SebKuzminsky, 2017-02-12:
The joints_axes branch has been merged into master, and has received bug fixes there.  Please try this with the master branch instead of joints_axes and let us know if the problem persists.

SebKuzminsky, 2017-02-12:
I have deleted all the old versions of the joints_axes branches to avoid this kind of confusion in the future...

grandixximo, 2017-02-12:
Ok i tried with the master branch, the master branch does not freeze, but still will not let me switch mode to joint to jog out of limit.

this is what the command line writes when i try to switch to joint mode

USRMOT: ERROR: command timeout
task: main loop took 1.010121 seconds

grandixximo, 2017-02-12:
Ok, i tested again and now it's not happening O.0
I did not do anything to the code at all

SebKuzminsky, 2017-02-13:
That USRMOT timeout indicates exceedingly poor realtime performance, which makes everything terrible.  Are you running this test in a virtual machine?  In any case I believe this is not a bug in LinuxCNC.

closed 2017-02-13 by SebKuzminsky