LinuxCNC/linuxcnc #232: halcompile --document

jethornton, 2017-02-14:
When testing halcompile --document with C style multi-line comments I get an error:
 ~/linuxcnc/components $ halcompile --document stmbl.comp
stmbl.comp:3:1: Trying to find one of "$", "pin", "param", "function", "variable", "option", "see_also", "notes", "description", "license", "author", "include", "modparam"
>  */ simple example of a realtime comp that does networking/*
>  ^
while parsing File():

jepler, 2017-02-14:
Attach the affected comp file.

Based on looking at the un-marked-up text of your original submission,
~~~~
 ~/linuxcnc/components $ halcompile --document stmbl.comp
stmbl.comp:3:1: Trying to find one of "$", "pin", "param", "function", "variable", "option", "see_also", "notes", "description", "license", "author", "include", "modparam"
>  */ simple example of a realtime comp that does networking/*
>  ^
while parsing File():
~~~~
it looks like you may have the delimiters of a C/C++ comment confused: They begin with `/*` and end with `*/` but you have it the other way around.

I tested by modifying one of the comp files in the testsuite, adding the comment on the first line:
~~~~
/* Test that names match */
component names_match;
license "GPL";
pin out bit out;
function _ nofp;
;;
FUNCTION(_) {}
~~~~
and didn't encounter any problems with `halcompile --document`.

jethornton, 2017-02-14:
faulty example where the comment markers were backwards.

closed 2017-02-14 by jethornton