LinuxCNC/linuxcnc #233: 5 axis wrong preview with W axis

grandixximo, 2017-02-16:
## Here are the steps I follow to reproduce the issue:

 1.Open the 5axis bridgemill configuration
 2.Home 
 3.Move W axis

## This is what I expected to happen:

The tool in the axis graphical preview should not move, because X and Y position of the tool tip did not move.

## This is what happened instead:

The tool in the axis graphical preview moves and plots a movement while it should not, also after moving W, the B axis orientation will plot a movement of the tool tip, but B axis should only change the orientation of the tool around the tip, it should plot any movement of the tip of the tool, the correct behavior can be observed when W is in position 0

You can also observe the difference in movement previews from vismach to axis, if W is set to 50 and B and C are 45, jogging B and C has different effects in the previews, the correct behavior is the vismach behavior.

## It worked properly before this:

Haven't tested

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): master simulation
 * I am using this kernel version (shown by `uname -a`): 4.4.0-62-generic
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [x] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):2.8.0~pre1
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):axis
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): none

luzpaz, 2023-06-26:
Is this still reproducible for you @grandixximo  ?

grandixximo, 2023-06-27:
Yes, the vismatch is also "incorrect" now.
But I guess the 5axis concept for linuxcnc is different from what I'm used to.
There is now an example code, and looks like W is used for drilling purposes, I've not seen a CAM that would manage working 5 axis like that.
How I use it W is loaded as tool length, the toolpath are XYZBC, W tool length is loaded with G43 and sent to W0 to correctly compesate for tool length.

grandixximo, 2023-06-27:
A system to define what the extra axis are supposed to represent to be defined in the INI file for [DISPLAY] would be nice, as after ABC the UVW can be used in different manners by different machines.

Sigma1912, 2023-06-27:
I gather that the opengl code used in vismach and preview is no longer up to date with current developments in 3D rendering and may need to be updated/replaced at some time.
Also note that the preview behavior seems to have changed since version 2.8 where uvw moves were rotated by ABC rotations but this does not seem to be the case since version 2.9. Now it's parallel to XYZ regardless of ABC rotation. One used to be able to customize the behavior somewhat by changing the order of the axis letters in the [Display] section but I could not get the rotation working.
I have worked on a sim config for a machine with a rotating nutating head that included IDENTITY, TCP and TOOL kinematics and also a crude remap for tilted work plane which I posted on the forum a little while ago.
Feedback was literally zero.

grandixximo, 2023-06-27:
> I gather that the opengl code used in vismach and preview is no longer up to date with current developments in 3D rendering and may need to be updated/replaced at some time. Also note that the [...]

Please post here the link.
I think the proper implementation would require a Tool Length column in the tool table, as it should not be axis related, need see what the devs think.

Sigma1912, 2023-06-27:
https://forum.linuxcnc.org/show-your-stuff/49103-kinematic-model-for-a-5axis-mill-with-universal-45-nutating-head#271334
Documentation is also provided in the zip folder.

grandixximo, 2023-06-27:
When you say TOOL kinematics work, what do yuou mean specifically?

Sigma1912, 2023-06-27:
When in Tool kinematic mode XYZ motion commands are executed in the current tool coordinate system. So a Z move should move the tool along its rotational axis (eg for drilling).
Motivation here was to create a sim that has TCP (where the CAM does the calcurations) and TOOL kinematics where required CAM functionality is done in LinuxCNC so prismatic machining with tool side rotation can be done without the need of a CAM system. The operator can easily program ,say a circular pocket in any tool orientation directly on the machine.
Note that 'tilted work plane' functionality comes 'for free' on machines with table rotation since here the Z axis remains parallel to the tool rotional axis. So this is only of interest on machines with tool side rotation like the featured rotating/nutating head or also the 'bridgemill' configuration.

Sigma1912, 2023-06-27:
I see this more of an exploration of using 'Tilted Work Plane' functionality. Further work would include:
- Since the feature requires basic CAM functionality to orient the tool to the requested work plane there needs to be a machine specific model to calculate the joint positions. Ideally this would be done automatically using a config file that then builds the model automaticcally.
-  The Python remap would need a lot more testing  and debugging. In the end this would really need to be implemented in the interpreter rather than a python remap.
-  To make things clearer to the user an augmented preview would be preferable. I implemented some of my ideas in the custom vismach patch but it would be much better to actually fuse the gcode preview and the vismach model into one.

grandixximo, 2023-06-27:
> When in Tool kinematic mode XYZ motion commands are executed in the current tool coordinate system. So a Z move should move the tool along its rotational axis (eg for drilling) whatever the tool orientation. Motivation here was to create a sim that has TCP (where the CAM does the calcurations) and TOOL kinematics where required CAM functionality is done in LinuxCNC so prismatic machining with tool side rotation can be done without the need of a CAM system. The operator can easily program ,say a circular pocket in any tool orientation directly on the machine. Note that 'tilted work plane' functionality comes 'for free' on machines with table rotation since here the Z axis remains parallel to the tool rotional axis. So this is only of interest on machines with tool side rotation like the featured rotating/nutating head or also the 'bridgemill' configuration.

Had to read it a couple of times, but I think I understand, you switch between TOOL kins and TCP kins with G12.1 Pxx correct?

grandixximo, 2023-06-27:
To return on topic yes, original post was about the 5axis mill preview not being correctly represented, there are many paths devs could take to fix that.
But looking at the 5axis bridgemill configuration on 2.10 it's become something that cannot help the user understand 5 axis machining on a machine like that, and the "teleporting" is dangerous.

andypugh, 2023-06-27:
> My understanding of W was that it represented tool length. In the toolpaths generated by CAM software for 5-axis XYZBC movements, tool length is accounted for by loading the appropriate W axis offset from the tool table.

This can be done, but isn't necessarily correct. 

A common interpretation is for XYZ to represent a position in cartesian space, and then A and B to represent rotations about that point that determine the tool direction. 

In some interpretations UVW are then moves in a reference frame aligned with the tool axis, so that W would be a drilling move along the tool axis, and U and V would set a position in that plane (Where do U and V point? This would be implementation specific. (ie whatever the kinematics module in use chooses to do) 

Tool length is not generally the W _axis_ but is, rather an offset _along_ the tool axis (which is parallel to W). This is often passed into the kinematics on a HAL pin. (And can actually be pulled out of any column of the tool table, depending on the whim of the system integrator)

Sigma1912, 2023-06-27:
> switch between TOOL kins and TCP kins with G12.1 Pxx correct?

Kinematic switching is done using M428, M429 and M430.

grandixximo, 2023-06-28:
> This can be done, but isn't necessarily correct.
>
> A common interpretation is for XYZ to represent a position in cartesian space, and then A and B to represent rotations about that point that [...]

How should tool length be managed? You think current 5axis bridgemill is a correct representation?