LinuxCNC/linuxcnc #235: Adding pycam to the LinuxCNC organisation?

njh, 2017-02-16:
Hello,

There has been a little burst of activity on bringing the development of pycam back to life:

https://sourceforge.net/p/pycam/mailman/pycam-devel/

It has been proposed to move development to Github but someone already has the org 'pycam':

https://github.com/pycam/

Do you think it would be reasonable to move the repository to the LinuxCNC organisation as a parent project?

https://github.com/linuxcnc/pycam

The two projects have a lots of users and developers in common.

nick.

andypugh, 2017-02-21:
I am not sure it makes sense to move the cross-platform PyCAM project under a dedicated "Linux" umbrella. (This would have been less of a problem if we had been allowed to keep EMC2 as the name).

njh, 2017-02-21:
Well right now nobody can get the current version to run on anything apart from Linux!
But hopefully that will be fixed.

While it is possible to do VCarve → LinuxCNC and PyCam → Mach3, the open source fans are more likely to do PyCam → LinuxCNC.

I personally see PyCAM as being part of an Open Source toolchain. And LinuxCNC being more synonymous with "Open Source CNC". Realtime Kernel aside - it should be able to run on other Unixes - just not worth the effort?

SebKuzminsky, 2017-02-21:
I also think of PyCAM and LinuxCNC as both being part of the open source fabrication toolchain/ecosystem (along with FreeCAD and SolveSpace).  And just like LinuxCNC is not necessarily limited to Linux, PyCAM is not necessarily limited to Python (for example see SebKuzminsky/pycam@99415b22d).

LinuxCNC has had a couple of folks work on and off on porting LinuxCNC to FreeBSD, most recently Edward Napierala.  There's no deep technical reason why our "uspace" aka "sim" system couldn't be made to run on FreeBSD, just no one has gotten all the way through it yet.

I think our slightly inaccurate names are no obstacle to working together.  I think any reluctance from the LinuxCNC side stems from the devs feeling overworked already, and not wanting to take on additional responsibilities for a new project.  Part of the problem is that we're unsure what it would actually *mean* to have the PyCAM Project be part of the LinuxCNC Organization. 

I propose we shelve this discussion for a while.

trasz, 2017-10-11:
FWIW, PyCAM has an official FreeBSD port/package (https://www.freshports.org/search.php?query=pycam&search=go&num=10&stype=name&method=match&deleted=excludedeleted&start=1&casesensitivity=caseinsensitive)

SebKuzminsky, 2017-10-11:
Neat!  @trasz, have you tried the recent new releases, 0.6.0 and 0.6.1?  And hopefully there's a 0.6.2 coming out in the not too distant future...

trasz, 2017-10-12:
I've tried, and it... well, segfaults.  Python27 segfaults, to be exact, just after showing the window with a menu bar.  Any suggestions on how to debug it?

SebKuzminsky, 2017-10-12:
Please open an issue in the [pycam repo](https://github.com/sebkuzminsky/pycam) and let's discuss it there.  Be sure to specify the version you're using clearly.

luzpaz, 2023-06-26:
[image]

Looks like development stopped 2 years ago