LinuxCNC/linuxcnc #238: hm2_7i90 and hm2_7i43 drivers fail silently with null config string

pcw-mesa, 2017-03-02:
## Here are the steps I follow to reproduce the issue:
(PC with 7I90 connected to working EPP port at default 0x378 address)
 1. halrun
 2. halcmd: loadrt hostmot2
 3. halcmd: loadrt hm2_7i90
 4. halcmd: show pin

## This is what I expected to happen:
show 7i90 pins (even without hm2_7i90 config string, defaults should be ok)
## This is what happened instead:
No pins, hm2-7i90 loads without error but never probes hardware
hm2_7i43 has the same issue
a single space in the config string will fix the problem, that is, config="" fails
but config=" " works as expected
## It worked properly before this:
Not sure, may always have been this way
## Information about my hardware and software:
 Same issue LinuxCNC 2.7, 2.7.7 ,2.7.8, master

andypugh, 2017-03-06:
Is this simply due to this bit of code? 

    for (i = 0; i < HM2_7I43_MAX_BOARDS; i ++) {
        if(!config[i] || !*config[i]) break;

If there is no config string, it just exits.

SebKuzminsky, 2017-03-06:
Yeah that looks pretty suspicious.

andypugh, 2017-03-07:
It looks like hm2_register does exactly the right thing with a null configuration string, so I rather suspect we can simply ignore that test. 
It isn't immediately obvious how you would pass a blank config to more than one board. But that's a corner-case. 
I sold my 7i43-powered machine, so I can't actually test the change.

andypugh, 2017-04-04:
@pcw-mesa Are you able to test this change?

pcw-mesa, 2017-04-04:
Yes, I can test this later today.

andypugh, 2017-08-08:
PCW. Did you test this change?

pcw-mesa, 2017-08-08:
Ack I've forgotten if I did or not, My test LinuxCNC CPU with a parallel port is still in a box after moving
I'll dig it out and try it again

pcw-mesa, 2017-08-08:
The error is still there (todays master) at least with the 7i90

andypugh, 2017-08-08:
Nothing has been changed yet. Are you in a position to compile LinuxCNC?

pcw-mesa, 2017-08-08:
yes

andypugh, 2017-08-08:
Can you try it after deleting the "  if(!config[i] || !*config[i]) break;" and then recompiling? 
I think we need to test that an empty config string does start the card, and that a non-emptly one still works.

pcw-mesa, 2017-08-08:
config[i] no longer exists in the current source

andypugh, 2017-08-10:
I just pushed a fix to the 2.7 branch. It would be good to test with actual hardware, but I am confident that it works.

pcw-mesa, 2017-08-10:
I'll test this either later today or tommorow

pcw-mesa, 2017-08-11:
Well it loads the driver now with a null config string but goes off into the weeds, trying to access a second card (the 7I90.1) (with a config string or not):
 

hm2/hm2_7i90.0:     IO Pin 070 (P3-45): IOPort
hm2/hm2_7i90.0:     IO Pin 071 (P3-47): IOPort
hm2/hm2_7i90.0: registered
hm2_7i90.0: board at (ioaddr=0x0378, ioaddr_hi=0x0778, epp_wide ON) found
Linux parallel port @632 not found
hm2_7i90: failed to clear EPP Timeout!
hm2: no firmware specified in config modparam!  the board had better have firmware configured already, or this won't work
hm2_7i90.1: EPP timeout on data cycle of read(addr=0x0100, size=4)
hm2_7i90: failed to clear EPP Timeout!
hm2/hm2_7i90.1: error reading hm2 cookie
hm2_7i90.1: board at (ioaddr=0x0278, ioaddr_hi=0x0000, epp_wide ON) not found!
hm2_7i90.0: releasing board
hm2/hm2_7i90.0: unregistered
hm2_7i90: rtapi_app_main: Input/output error (-5)hm2/hm2_7i90.0:     IO Pin 070 (P3-45): IOPort
hm2/hm2_7i90.0:     IO Pin 071 (P3-47): IOPort
hm2/hm2_7i90.0: registered
hm2_7i90.0: board at (ioaddr=0x0378, ioaddr_hi=0x0778, epp_wide ON) found
Linux parallel port @632 not found
hm2_7i90: failed to clear EPP Timeout!
hm2: no firmware specified in config modparam!  the board had better have firmware configured already, or this won't work
hm2_7i90.1: EPP timeout on data cycle of read(abut now with a config string or not it goes off into the weeds, trying to access a second card (the 7I90.1):ddr=0x0100, size=4)
hm2_7i90: failed to clear EPP Timeout!
hm2/hm2_7i90.1: error reading hm2 cookie
hm2_7i90.1: board at (ioaddr=0x0278, ioaddr_hi=0x0000, epp_wide ON) not found!
hm2_7i90.0: releasing board
hm2/hm2_7i90.0: unregistered
hm2_7i90: rtapi_app_main: Input/output error (-5)
<stdin>:2: waitpid failed /home/pcw/linuxcnc-dev/bin/rtapi_app hm2_7i90
<stdin>:2: /home/pcw/linuxcnc-dev/bin/rtapi_app exited without becoming ready
<stdin>:2: insmod for hm2_7i90 failed, returned -1

<stdin>:2: waitpid failed /home/pcw/linuxcnc-dev/bin/rtapi_app hm2_7i90
<stdin>:2: /home/pcw/linuxcnc-dev/bin/rtapi_app exited without becoming ready
<stdin>:2: insmod for hm2_7i90 failed, returned -1

andypugh, 2017-08-11:
Do you have a port address of 0x632 specified, or is that bogus?

pcw-mesa, 2017-08-11:
Its bogus

pcw-mesa, 2017-08-11:
That is, with a null config string it looks like it finds the (one) 7I90 I have connected
at the Default 0x378 address, then goes on to try the next standard parallel port address
(0x278 = 632). Master works as expected (and still silently fails when no config string is present)

andypugh, 2017-10-10:
I have pushed a new fix. It works here with both a single 7i43 and a single 7i90. 
I don't see any reason for it not to work with multiples of either, but lacked the hardware to check.

closed 2017-10-10 by andypugh