LinuxCNC/linuxcnc #239: interp's named variables for tool info are wrong

SebKuzminsky, 2017-03-03:
Forum user jtc reports a bug with #<_current_pocket>:

https://forum.linuxcnc.org/10-advanced-configuration/32412-bug-in-current-pocket#88904

The [documentation for #<_current_pocket>](http://linuxcnc.org/docs/devel/html/gcode/overview.html#sub:system-parameters) says:

```
#<_current_pocket> - Return pocket number of the current tool.
```

However, on startup the value is 1, which is wrong.  And after Tx M6 the variable holds the index into Interp's tool table, which may differ from the pocket number in the tool table file and thus is also wrong.

To further complicate the issue, Random and Non-random tool changes differ in their behavior here.

SebKuzminsky, 2017-03-06:
The branch "2.7-tool-info" adds a test that demonstrates the current bogus behaviors.

rene-dev, 2018-02-25:
https://github.com/LinuxCNC/linuxcnc/blob/c8a718ea374289d661fbf5239096de53eddd3975/src/emc/rs274ngc/interp_namedparams.cc#L664
this returns the index number, not the pocket number.

rene-dev, 2020-04-20:
fixed in 2.8 and master by dab2a051b30e600ef95fbe911053386ef44631d0

closed 2020-04-20 by rene-dev