LinuxCNC/linuxcnc #241: unexpected move after program stop when remap on_abort is used

propcoder, 2017-03-04:
Sorry, not enough time to describe everything in detail.
## Here are the steps I follow to reproduce the issue:
TBD
## This is what I expected to happen:
Normal stop.
## This is what happened instead:
unexpected move after program stop when remap on_abort is used

## Information about my hardware and software:
TBD
 * I am using this Linux distribution and version: Debian Wheezy x86
 * I am using this kernel version: TBD
 * I am running LinuxCNC 2.7.8
   * [+] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)

alexfloca, 2017-03-27:
hi, reported this a couple of years ago, i have just disabled the  ON_ABORT_COMMAND

KMevada, 2018-01-07:
Even I got the same issue while using ON_ABORT_COMMAND
It seems that vector chained_points  in emccannon is holding some point and not clearing it out. When we call ON_ABORT_COMMAND this generates LINEAR move msg in flush_segments().

Did u find any solution to it?

KAUSHIk

closed 2023-10-03 by andypugh