LinuxCNC/linuxcnc #245: halcompile bug in 2.7 with singleton userspace components

SebKuzminsky, 2017-03-19:
User Trần Ngọc Quân posted a singleton userspace comp on the emc-users list: https://sourceforge.net/p/emc/mailman/emc-users/thread/080135e6-4df5-ff80-7c7f-849e9738d966%40gmail.com/#msg35732613

This comp has a few bugs, but also exposes a bug in halcompile: singleton userspace components don't work.  The user_mainloop() function uses the per-instance macros to access the hal pins, but __comp_inst isn't defined.  The docs say to use FOR_ALL_INSTS() in user_mainloop(), but that's silly for a singleton comp.  Maybe singleton comps should define __comp_inst to be __comp_inst_first?

vnwildman, 2017-03-19:
My comp build and run ok (after some fix) on raspbberypi 3 with machinekit. [image]

[cpu_thermalusr.txt](https://github.com/LinuxCNC/linuxcnc/files/853073/cpu_thermalusr.txt)

jepler, 2017-03-19:
@SebKuzminsky  if we take your proposed change, can we still manage to write FOR_EACH_INST in a way that userspace singleton components that used it "unnecessarily" will still work?

andypugh, 2021-02-19:
This apparently got fixed in Master, but not in 2.7 or 2.8?
https://github.com/LinuxCNC/linuxcnc/commit/2caf046d836cd0d0dbac25cb7e8d2f5c863f1d40#diff-4068db2aca22e5eb7718db692852b257d830760c7e4ed31b371591733070abdf

jepler, 2021-02-19:
Closing as #664 resolved this in Master.  Release Managers can cherry-pick the fix in #664 back to stable versions if desired.  Thanks!

closed 2021-02-19 by andypugh