LinuxCNC/linuxcnc #249: Code cleanup ... Stepconf Wizard

WJHildreth, 2017-03-31:
Hello,

I think there is some old code laying about in stepgen wizard that may need cleaned out.

I am running a Gantry router that was configured with stepconf.  In an effort to learn more about the system and its software I compared my machines.ini file entries to those that are documented in the INI Configuration documentation.  There are a few undocumented variables I came across and after consulting the emc-users mailing list was told that they are probably leftover from the old days.

Here are the sections and variables in question:

[EMCMOT] section
COMM_TIMEOUT = 1.0
COMM_WAIT = 0.010

[TRAJ] section
CYCLE_TIME = 0.010

I have commented these out in my INI file to no ill effect that I am aware of.  If these variables are no longer required I think they should be removed from the stepconf wizard when it writes out it's configuration files.

Here are some specifics about my setup.

OS: Debian Wheezy installed from the Hybrid ISO
LinuxCNC Version 2.6.7
GUI: Axis
Machine: XYZ Stepper driven Gantry Router
Config: Basic config generated from stepconf. Axes only, no limit, home switches, no spindle control, no external E-Stop.  Stepper drivers are driven through parallel port to CW230 Stepper controllers.

Thank you for your time a patience,

Joe Hildreth

nicokid, 2017-04-03:
Stepconf currently writes  some fixed values in these sections:
> [EMCMOT]
> COMM_TIMEOUT = 1.0
> COMM_WAIT = 0.010

> [TRAJ] 
> CYCLE_TIME = 0.010

But I do not know if this stuff is still used by LinuxCNC. I think we need the opinion of some other person.
Nicola.

SebKuzminsky, 2017-07-14:
[EMCMOT]COMM_TIMEOUT is the number of seconds that Task will wait for Motion to acknowledge receipt of a message.  Defaults to 1.0 seconds.  This is an important ini variable and t is a bug that this is not documented.

[EMCMOT]COMM_WAIT is not used and should be removed from all our configs and config-writer programs like stepconf and pncconf.  It's been in our code base since the beginning, but it has never been used by our code.

[TRAJ]CYCLE_TIME is not used and should be removed from all our configs and config-writer programs like stepconf and pncconf.  The trajectory planner runs (as part of Motion) in the so-called "servo" thread, and its cycle time is computed automatically based on the servo thread period.  There is a bunch of fossil remains of this variable in our code and docs, removing it will be a bit involved.

andypugh, 2017-07-14:
On 14 July 2017 at 19:28, Sebastian Kuzminsky <notifications@github.com>
wrote:

[TRAJ]CYCLE_TIME is not used and should be removed from all our configs and
config-writer programs like stepconf and pncconf. The trajectory planner
runs (as part of Motion) in the so-called "servo" thread, and its cycle
time is computed automatically based on the servo thread period. There is a
bunch of fossil remains of this variable in our code and docs, removing it
will be a bit involved.

It is probably fairly easy to persuade the config-updater script to delete
it (or replace it with a "we removed this" comment.

-- 
atp

SebKuzminsky, 2017-07-14:
@andypugh That's a good idea.  I favor just removing it.

SebKuzminsky, 2017-07-14:
I added documentation for [EMCMOT]COMM_TIMEOUT in 2.7 and master (140104aad62e21da509953f4febe26a2eeb1c3ff), and I removed [EMCMOT]COMM_WAIT in master but left it in 2.7 (945de033f9bee05c311dd95a64793c2efcd43b79, b167ec2d8756f3e891417674a6d90ad6fdcbb8fe).

SebKuzminsky, 2017-07-15:
I removed [TRAJ]CYCLE_TIME in master (a5c7736af80bb4e357b4db4ec126c6347f5e999f) but left it in 2.7.

@andypugh I did not teach update_ini to drop [EMCMOT]COMM_WAIT or [TRAJ]CYCLE_TIME, though as you suggested, it should.

I opened #306 to track some follow-on work that I discovered while working on this.

@WJHildreth thanks for pointing out this problem.

closed 2017-07-15 by SebKuzminsky