LinuxCNC/linuxcnc #256: linuxcnc does not start with mesa board

valeriob01, 2017-04-10:
## Here are the steps I follow to reproduce the issue:

 1. run pncconf and generate a config
 2. run linuxcnc
 3. freeze up

## This is what I expected to happen:
axis gui shows up
## This is what happened instead:
axis gui does not start

## Information about my hardware and software:

 * I am using this Linux distribution and version:
~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 7.9 (wheezy)
Release:	7.9
Codename:	wheezy

 * I am using this kernel version:
~$ uname -a
Linux cncmk1 3.4-9-rtai-686-pae #1 SMP PREEMPT Debian 3.4.55-4linuxcnc i686 GNU/Linux

 * I am running ...
   * [X ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version 2.7.8

 * I am using this user interface AXIS
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): 

Mesa 6i25-7i76

andypugh, 2017-04-10:
I am not sure that this is a LinuxCNC bug as such. This seems likely to be a problem specific to your particular PC. 
What happens if you edit the INI file to use tklinuxcnc rather than axis in the [DISPLAY] section?

valeriob01, 2017-04-10:
I did the change and nothing shows, it is frozen again: 
[image]

What do I have to do after physically installing the mesa board? install some driver ?

andypugh, 2017-04-10:
The Mesa drivers are built-in to LinuxCNC, no extra config should be needed. 
Can we move this to the forum?

valeriob01, 2017-04-10:
Yes, already done, but now I have a trace:

[image]

SebKuzminsky, 2017-04-10:
@valeriob01 reports in email that they fixed the configuration problem and things work now:

> Followup: the issue was in the servo jitter value. Looking at dmesg
> there are a consistent number of messages indicating that one cpu was
> stuck, and other messages about hm2. Now I have left the default
> untouched and Axis starts.

closed 2017-04-10 by SebKuzminsky