LinuxCNC/linuxcnc #268: Axis: can't change jog speed from keyboard shortcuts

micges, 2017-05-12:
## Here are the steps I follow to reproduce the issue:

1. Run axis sim config
2. Try to change jog speed from keybard

## This is what I expected to happen:

Jog speed changes.

## This is what happened instead:

Jog speed doesn't change.

## It doesn't work in:

2.6.13
2.7.7

jepler, 2017-06-09:
We discussed this on IRC but didn't reproduce the problem.  Please be more specific in your steps to reproduce.  Could it be related to a non-US keyboard layout?

sphericos, 2017-11-14:
I have noticed the same issue. I have installed a functional system using the stretch iso from here http://www.linuxcnc.org/testing-stretch-rtpreempt/linuxcnc-stretch-uspace-amd64-r9.iso everything was updated and LinuxCNC is version 2.7.-11
 
After restoring the configuration from another computer, everything functioned as expected with the exception of the jog speed shortcut keys. I assumed that it was a problem with my setup, config or the stretch iso so I followed the guide to build the Linux Mint version including compiling the same version of LinuxCNC from source. Again everything worked as expected except for the jog speed shortcut keys.

jepler, 2017-12-10:
We believe this was fixed by the commits listed above.

closed 2017-12-10 by jepler