LinuxCNC/linuxcnc #269: G-Code error when attempting to touchoff.

phillc54, 2017-05-15:
**## Here are the steps I follow to reproduce the issue:**
1. run axis sim
2. E-Stop off
3. Power on
4. press Home All
5. press Touch Off
6. press OK

**## This is what I expected to happen:**
Zero X axis in G54 coordinate system

**## This is what happened instead:**
The following error message:
    G-Code error in axis.ngc
    Near line 0 of
    /home/phill/blah/blah/blah/axis.ngc:
    Line with G10 does not have L1, L10, L11, L2, L20

**## Information about my hardware and software:**
 * Debian 7.11 (wheezy) in a VMware Fusion virtual machine:
 * 3.4-9-rtai-686-pae #1 SMP PREEMPT Debian 3.4.55-4linuxcnc
 * v2.8.0-pre1-3125-gd63aca6 as a rip build
 * AXIS gui
 * sim configs only - axis, lathe, historical_lathe, gantry, ldelta and rdelta all behave the same

Issue first appears in commit 001eb00, last working commit was 596363b.

SebKuzminsky, 2017-05-15:
I confirm this happens on the tip of master (d63aca6e).

TurBoss, 2017-05-15:
I'm on master too and the "regular offset" button up to "touch off" causes this problem too

Linux cnc28 3.4-9-rtai-686-pae #1 SMP PREEMPT Debian 3.4.55-4linuxcnc i686 GNU/Linux

The following error message:
G-Code error in axis.ngc
Near line 0 of
/path/to/file.ngc:
Line with G10 does not have L1, L10, L11, L2, L20

jepler, 2017-05-15:
@phillc54 thanks for the detailed bug report.  based on this, it's pretty clear I introduced the problem, so I'll try to resolve it soon.  For now you can probably just `git revert` the offending commit for local use.

jepler, 2017-05-15:
The problem stems from how after `G10 L2 ...`, `G10` appears in the list of "active gcodes" (`stat.gcodes[2]`); then the new code added at 001eb00313d28a872b275acbc29407f5266a0f42 tries to issue a `G10` code as a part of preview generation.

commit cb74ca8b (2017-05-15): axis: Fix reload of preview after touch off

.. see the added comment for more information

Close: #269
closed 2017-05-15 by cradek