LinuxCNC/linuxcnc #271: M code greater than 199:M15816

TurBoss, 2017-05-16:
## Here are the steps I follow to reproduce the issue:

 1. Open a very big program > 20 MB
 2. set G54 to any place
 3. run the program and stop it
 4. then open another program 

## This is what happened instead:

a erro is shown in a popup window :
[image]

## Information about my hardware and software:

linuxcnc latest iso with master rip install from github

Linux cnc28 3.4-9-rtai-686-pae #1 SMP PREEMPT Debian 3.4.55-4linuxcnc i686 GNU/Linux

Thanks

SebKuzminsky, 2017-05-16:
Please attach the G-code program to this issue.  (Apparently github supports 25 MB attachments: https://help.github.com/articles/file-attachments-on-issues-and-pull-requests/)

TurBoss, 2017-05-16:
ngc file attached generated with blendercam

[Piedra_Irregular_Muestra.ngc.zip](https://github.com/LinuxCNC/linuxcnc/files/1005936/Piedra_Irregular_Muestra.ngc.zip)

TurBoss, 2017-05-16:
I'm unable to reproduce this on the simulator x64 but always on the machine x86

pentium 4 2GB ram

SebKuzminsky, 2017-05-16:
I can reproduce this on the current tip of master (v2.8.0-pre1-3126), but it does not happen on 2.7.8.  I'm testing on a 32-bit x86 VM installed from our 2.6 or 2.7 Live/Install ISO.

On both, the active mcodes as reported by Status, both immediately after aborting the big program and also after trying to load 3D_Chips.ngc are: "M3 M9 M48 M53 M0".

jepler, 2017-05-16:
I wonder if this is also introduced by the same commit that caused the "G10" problem.  Do we stuff something in the active M codes that is related to file position, and I'm erroneously trying to restore it?

jepler, 2017-05-16:
Yup.
~~~~
initcodes ['G20', 'g90', 't1 m6', 'g53 g0 X-2.74416465 Y-2.99650464 Z-0.60384252', 'G1.0', 'G17.0', 'G40.0', 'G21.0', 'G90.0', 'G94.0', 'G54.0', 'G43.0', 'G99.0', 'G64.0', 'G97.0', 'G91.1', 'G8.0', 'G43.1 X0.00000000 Y0.00000000 Z0.51100000', 'M3537', 'M3', 'M9', 'M48', 'M53', 'M0']
~~~~

I'll try to get this fixed later in the evening.

closed 2017-05-17 by cradek

TurBoss, 2017-05-17:
Thank you so much!

jepler, 2017-05-17:
@TurBoss you're welcome, and thank you for the actionable bug report.  And I'm sorry again for introducing these problems!  They are related to a feature I added that improves preview of "ungrounded" g-code programs, ones that depend on the current machine position and modal codes, rather than establishing them all at the beginning of the part program.

jarkko-hautakorpi, 2018-01-13:
How should the g code be modified to get rid of the error? Reinstall whole LinuxCNC?

commit 5b7892be (2017-05-17): axis: Fix reload of preview after abort

.. see the added comment for more information

Close: #271