LinuxCNC/linuxcnc #273: Axis: File is not closed after reading

laurivosandi, 2017-05-19:
## Here are the steps I follow to reproduce the issue:

 1. Open a file with Axis, run lsof on the opened file
 2. Open another file closing the first one, run lsof on the previously opened file
 3. Axis is still listed as using the first file

I realise that for large designs it might make sense to read G-code from the file directly, however considering computer resources nowadays there could be at least a switch to toggle the behaviour.

## This is what I expected to happen:

File is immediately closed after reading or at least closed when I open another file.

## This is what happened instead:

File is left open, resulting in locked files on CIFS shared folder.

## Information about my hardware and software:

 * I am using latest Git revision compiled on Fedora, same issue was in the LinuxCNC distro installed in the summer of 2016. I can check the version next time I'll be at the lab.

jepler, 2017-05-19:
You can probably at least prevent these CIFS files from being locked by using a program filter of "cat", e.g., if you consistently use the ".gcode" extension, find the `[FILTER] section of your inifile and add `gcode = cat`.  This basically causes AXIS to take a copy of the file in /tmp before opening it.

However, this may also affect how "reload" functions, I would have to review the source code to check that.

note: this is just a possible workaround, and what you describe still sounds like a bug that ought to be fixed.

laurivosandi, 2017-05-19:
Hello, which inifile are  you talking about? The documentation doesn't mention *where* the files are...

jepler, 2017-05-19:
`~/linuxcnc/configs` or `configs` in the source tree.

laurivosandi, 2017-05-19:
Machine configuration directory and file is dumped when LinuxCNC GUI is started up, it happened to be in this case ``/home/$USER/linuxcnc/configs/sim/axis/axis.ini``, I added ``ngc = cat`` in the ``[FILTER]`` section and no more the files are locked. Thanks for the workaround, btw this won't make a copy to ``/tmp``,  it just uses the ``cat`` command to read the file and cat just happens to close file properly after dumping the output to ``stdout``.

However without the workaround I think opening another file should definitely close the previous ones. It's essentially leaking memory and file handles as it is right now.

PR #299 (2017-07-14): gcodemodule: Make interp really close part program
For valid reasons (something to do with calling O-words from MDI),
the interpreter leaves the last file open even when close() is
called.  However, open() actually does close the old file before
opening a new one.

This three step process ensures the old file is closed,
no new file is opened, and the interpreter's internal state
is changed to "no file open"

@laurivosandi can you let me know if this fixes your issue?

Closes: #273
commit 4f0de2ae (2017-07-14): gcodemodule: Make interp really close part program

For valid reasons (something to do with calling O-words from MDI),
the interpreter leaves the last file open even when close() is
called.  However, open() actually does close the old file before
opening a new one.

This three step process ensures the old file is closed,
no new file is opened, and the interpreter's internal state
is changed to "no file open"

Closes: #273