LinuxCNC/linuxcnc #280: How to config the NML file

wilsonxiao, 2017-06-09:
How to config the NML file?
I want to run GUI in a networked computer to get emcstatus from another networked computer runing the realtime parts of emc2 and send the emcCommand to the runing emc2 computer. How should I to config the NML file or other files?

wilsonxiao, 2017-06-09:
Should I create the share files folder with the two computers?

jepler, 2017-06-09:
Nobody who uses LinuxCNC like this has ever contributed documentation of how to set it up (if such people even exist).  If you do, please consider submitting documentation about it via the github pull request interface.

There are also likely to be "missing pieces" such as how to open the same file for preview on the computer with the UI and on the computer where task is running.  We'd be happy to see improvements related to this submitted as pull requests as well.

For an example of a .ini / .nml combination we know works using nml-over-tcp (but only over the localhost), you can look in tests/linuxcncrsh-tcp in our master branch.

closed 2017-06-09 by jepler