LinuxCNC/linuxcnc #282: Test Axis of stepconf moves the axis when not told to

TurBoss, 2017-06-12:
## Here are the steps I follow to reproduce the issue:

 1. Launch stepconf and configure a machine with one or more axis/joints
 2. in the axis parameters page click on "Test Axis"
 3. Wait 30 secs or so

## This is what I expected to happen:
Nothing

## This is what happened instead:

The motor moves one step each 30 seconds

## It worked properly before this:
New install son no previous tests but in 2.8 from sources and from buildbot is OK

## Information about my hardware and software:
Debian GNU/Linux 7.11 (wheezy)
Linux cnc28 3.4-9-rtai-686-pae #1 SMP PREEMPT Debian 3.4.55-4linuxcnc i686 GNU/Linux

LinuxCNC 2.79 from the repo with Axis

my interface is a pcie parport with the WCH382

I'm not sure if is my parport card or a bug please confirm
Thanks

TurBoss, 2017-06-12:
```
turboss@turbo-box:~$ ssh jauria@192.168.10.136
jauria@192.168.10.136's password: 
Linux lcnc 3.4-9-rtai-686-pae #1 SMP PREEMPT Debian 3.4.55-4linuxcnc i686

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Tue Jun 13 01:56:21 2017 from turbo-box-3.local
jauria@lcnc:~$ halcmd show all
Loaded HAL Components:
ID      Type  Name                                            PID   State
    11  User  halcmd4259                                       4259 ready
     9  RT    __slow                                                ready
     8  RT    __fast                                                ready
     7  RT    threads                                               ready
     3  User  halcmd4154                                       4154 ready
     6  RT    hal_parport                                           ready
     5  RT    stepgen                                               ready
     4  RT    steptest                                              ready

Component Pins:
Owner   Type  Dir         Value  Name
     8  s32   OUT          5088  fast.time
     6  bit   IN          FALSE  parport.0.pin-01-out
     6  bit   IN          FALSE  parport.0.pin-02-out <== step
     6  bit   IN          FALSE  parport.0.pin-03-out <== dir
     6  bit   IN          FALSE  parport.0.pin-04-out
     6  bit   IN          FALSE  parport.0.pin-05-out
     6  bit   IN          FALSE  parport.0.pin-06-out
     6  bit   IN          FALSE  parport.0.pin-07-out
     6  bit   IN          FALSE  parport.0.pin-08-out
     6  bit   IN          FALSE  parport.0.pin-09-out
     6  bit   OUT         FALSE  parport.0.pin-10-in
     6  bit   OUT         FALSE  parport.0.pin-10-in-not
     6  bit   OUT         FALSE  parport.0.pin-11-in
     6  bit   OUT         FALSE  parport.0.pin-11-in-not
     6  bit   OUT         FALSE  parport.0.pin-12-in
     6  bit   OUT         FALSE  parport.0.pin-12-in-not
     6  bit   OUT         FALSE  parport.0.pin-13-in
     6  bit   OUT         FALSE  parport.0.pin-13-in-not
     6  bit   IN          FALSE  parport.0.pin-14-out
     6  bit   OUT         FALSE  parport.0.pin-15-in
     6  bit   OUT         FALSE  parport.0.pin-15-in-not
     6  bit   IN          FALSE  parport.0.pin-16-out
     6  bit   IN          FALSE  parport.0.pin-17-out
     6  s32   OUT             0  parport.0.read.time
     6  s32   OUT           228  parport.0.reset.time
     6  s32   OUT          4716  parport.0.write.time
     6  s32   OUT             0  parport.read-all.time
     6  s32   OUT             0  parport.write-all.time
     9  s32   OUT          1236  slow.time
     5  s32   OUT           593  stepgen.0.counts
     5  bit   OUT         FALSE  stepgen.0.dir ==> dir
     5  bit   IN           TRUE  stepgen.0.enable
     5  float IN       29.63179  stepgen.0.position-cmd <== cmd
     5  float OUT      29.63179  stepgen.0.position-fb ==> fb
     5  bit   OUT         FALSE  stepgen.0.step ==> step
     5  s32   OUT           216  stepgen.capture-position.time
     5  s32   OUT           432  stepgen.make-pulses.time
     5  s32   OUT           564  stepgen.update-freq.time
     4  float IN             15  steptest.0.amplitude
     4  s32   IN              0  steptest.0.dir
     4  bit   IN          FALSE  steptest.0.jog-minus
     4  bit   IN          FALSE  steptest.0.jog-plus
     4  float IN              0  steptest.0.maxaccel
     4  float IN             10  steptest.0.maxvel
[... 268 lines cut ...]
Owner   CodeAddr  Arg       FP   Users  Name
 00006  f8566000  f8bc3210  NO       0   parport.0.read
 00006  f85662a0  f8bc3210  NO       1   parport.0.reset
 00006  f85663a0  f8bc3210  NO       1   parport.0.write
 00006  f85660f0  f8bc3210  NO       0   parport.read-all
 00006  f8566570  f8bc3210  NO       0   parport.write-all
 00005  f8c2d350  f8bc3130  YES      1   stepgen.capture-position
 00005  f8c2d000  f8bc3130  NO       1   stepgen.make-pulses
 00005  f8c2d430  f8bc3130  YES      1   stepgen.update-freq
 00004  f83cc000  f8bc30c8  YES      1   steptest.0

Realtime Threads:
     Period  FP     Name               (     Time, Max-Time )
     999710  YES                  slow (      948,     3792 )
                  1 stepgen.capture-position
                  2 steptest.0
                  3 stepgen.update-freq
      99971  NO                   fast (     4944,    15480 )
                  1 stepgen.make-pulses
                  2 parport.0.write
                  3 parport.0.reset

jauria@lcnc:~$ 

```

SebKuzminsky, 2017-06-12:
Note how in the first "show all", the stepgen.position-cmd and .counts are fairly large, because the user jogged.  In the second "show all", the user waited for the stepper to take a single un-commanded step (~30 seconds after the end of the jog) and the .position-cmd and .counts both show the step was commanded.

The bug seems to be that steptest doesn't go to 0 velocity after a jog ends.

TurBoss, 2017-06-13:
After testing I can confirm that this is fixed

upgraded no bug
downgraded hit bug again
upgraded and seems fixed

Thanks!

SebKuzminsky, 2017-06-14:
My fix introduced another problem.  I have a real fix, I'll pick push it later tonight.

TurBoss, 2017-06-14:
Hello

Where i can find the associated deb file
I'm searching here : 
http://buildbot.linuxcnc.org/dists/wheezy/scratch-rt/binary-i386/

Thanks

SebKuzminsky, 2017-06-14:
@TurBoss The fix is in 2.7 now, no need to use the scratch repo any more.  If you do a normal upgrade using the buildbot's normal deb archive and you get version v2.7.9-1-g63a2eca9cc or newer you'll get the fix.

The fix I pushed to 2.7 is a little different from the fix you tested.  This second attempt at the fix worked much better in my testing, i'd like to hear your feeback on it.

TurBoss, 2017-06-14:
after 30 mins nothing happened so this is fixed no other strange things shown

Thaks

closed 2017-06-14 by SebKuzminsky