LinuxCNC/linuxcnc #285: Inconsistent mode switching

pkmcnc, 2017-06-19:
## Here are the steps I follow to reproduce the issue:

 1. Start a config (axis_mm)
 2. Power on the machine
 3. Switch to MDI mode

## This is what I expected to happen:
halui.mode-is-mdi activates

## This is what happened instead:
halui.mode-is-mdi activates for a short moment and then halui.mode-is-manual is active again.
If MDI command is executed the mode switches to MDI.

Also, the machine starts in a manual mode then switches to a different mode for a brief moment, then switches back to manual (I have a relay connected to manual mode output and can hear that).

## It worked properly before this:
Not sure about 2.7

## Information about my hardware and software:

I am using Debian image.
I am running a binary I built myself from source.
I am using this LinuxCNC version v2.8.0-pre1-3169-gdf1c3ba

rene-dev, 2017-12-01:
seems to be axis related, cant reproduce this in gmoccapy.
works in 2.7 in axis.

rene-dev, 2017-12-13:
I think this can be closed., fixed by https://github.com/LinuxCNC/linuxcnc/commit/b7a13a899c59fb329602e2493d5edd07d3bf4244

closed 2017-12-13 by jepler