LinuxCNC/linuxcnc #293: Hit EStop while path loading freezes Axis after load is completed

TurBoss, 2017-06-27:
## Here are the steps I follow to reproduce the issue:

 1. Load a big program
 2. Hit the remote EStop while still loading

## This is what I expected to happen:

nothing

## This is what happened instead:

after the path is fully loaded axis goes to a frezzed state

## Information about my hardware and software:

 * I am using Debian GNU/Linux 7.11 (wheezy)
 * I am using Linux cnc28 3.4-9-rtai-686-pae #1 SMP PREEMPT Debian 3.4.55-4linuxcnc i686 GNU/Linux

 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ X] A binary I built myself 2.8.0pre
   * [ ] A binary version from some other source besides linuxcnc.org
 
* I am AXIS GUI

Thanks

jepler, 2017-06-28:
Can you give a little more explanation about how you reproduce this problem?

I used sim/axis/axis.ini for testing.

I created a part program that takes several seconds to load:
~~~~
g20 g64 g90
s3400 m3

g0 z1
g0 x2 y0
g1 z-.1 f24

o100 repeat [80000]
g91 g1 @-.0025 ^4.5
o100 endrepeat

g90 g0 z1
m2
~~~~
then I arranged two different sequences in a terminal window to trigger an estop condition:

- via iocontrol
~~~~
sleep .5; halcmd unlinkp iocontrol.0.emc-enable-in; halcmd setp iocontrol.0.emc-enable-in false; sleep .5; halcmd net estop-loop iocontrol.0.emc-enable-in
~~~~

 - via halui
~~~~
sleep .5; halcmd setp halui.estop.activate 1; sleep .5; halcmd setp halui.estop.activate 0
~~~~

however, neither sequence left the AXIS process unresponsive once it had finished loading the part program.

TurBoss, 2017-06-28:
will try to reproduce tomorrow

TurBoss, 2017-06-28:
I'm sorry but I'm unable to reproduce this again

so there is no bug

Thanks

closed 2017-06-28 by TurBoss