LinuxCNC/linuxcnc #297: fix gmoccapy gstreamer support on Debian Stretch

SebKuzminsky, 2017-06-29:
Gmoccapy in LinuxCNC 2.7 imports "gst" (from python-gst0.10), and presumably uses the helper libraries from gstreamer0.10-plugins-base.

As a result the LinuxCNC 2.7 debian packages depend on python-gst0.10 and gstreamer0.10-plugins-base.  These packages are available in Ubuntu Lucid and Precise, and Debian Wheezy and Jessie, but not in Debian Stretch.

Stretch has python-gst1.0 and gstreamer1.0-plugins-base instead, but we don't know if they work with Gmoccapy, or what changes might be needed.

For now we've simply removed (c98aa61970639b5c02b3f255f9398fd5fe9de3b5) the dependency on python-gst0.10 and gstreamer0.10-plugins-base on Stretch, but this probably breaks that part of gmoccapy.  

Someone needs to make the appropriate changes, if any, that gmoccapy needs to support both gstreamer 0.10 and gstreamer 1.0, then we can add the dependency on the new packages to our Stretch packages.

SebKuzminsky, 2017-06-29:
See also #298.

jepler, 2017-06-29:
I added gst to http://wiki.linuxcnc.org/cgi-bin/wiki.pl?MinimumSoftwareVersions.

saxa, 2017-06-29:
Excuse my ignorance, but what is gstreamer used for in LinuxCNC ?

jepler, 2017-06-29:
~~~~
$ git grep 'import gst' origin/2.7
origin/2.7:src/emc/usr_intf/gmoccapy/gmoccapy.py:            import gst
origin/2.7:src/emc/usr_intf/gmoccapy/player.py:import gst
origin/2.7:src/emc/usr_intf/gmoccapy/player.py:        import gst
origin/2.7:src/emc/usr_intf/gscreen/gscreen.py:    import gst
~~~~
I don't know, but gscreen and gmoccapy can (possibly optionally) import it.  I _think_ it has to do with showing input from a camera within the UI.

gmoccapy, 2017-07-17:
I made some tests and as far as I found, there is no problem to change from
python-gst0.10 to python-gst1.0 as the import statement remain the same.

IMHO we should just get rid of that dependency, as gscreen and gmoccapy uses that import only to play alarm or warning sounds.

If that module is not found, the sound capability of the GUI will be disabled. I just pushed a new version, where the user would get a information on the console, if the gst module is not found.

So no problem from my side to port to stretch.

jepler, 2017-07-17:
I can only test in an emulator right now, but I see that compared to the base system, python-gst-1.0 only uses 94.2kB disk space as reported by 'apt install'.  The default code refers to a file in sound-theme-freedesktop, which has an installed-size of 554kB.  It does appear that calling the 'run' method on a gmoccapy.player.Player instance does *something* because when I do it interactively it takes a moment to return to the Python ">>>" prompt.

so, given the small size of these files that are not in the base system, I think we should have linuxcnc depend on python-gst-1.0 and sound-theme-freedesktop.

While I'm at it, I notice there are some references to /usr/share/sounds/ubuntu which is not likely to work on any Debian OS.  Perhaps those should be fixed to refer to /usr/share/sounds/freedesktop ?  Would you like me to file issues about this @gmoccapy @c-morley ?

jepler, 2017-07-18:
Fixed by fe2483c

closed 2017-07-18 by jepler