LinuxCNC/linuxcnc #298: fix gscreen gstreamer support on Debian Stretch

SebKuzminsky, 2017-06-29:
Gscreen in LinuxCNC 2.7 imports "gst" (from python-gst0.10), and presumably uses the helper libraries from gstreamer0.10-plugins-base.

As a result the LinuxCNC 2.7 debian packages depend on python-gst0.10 and gstreamer0.10-plugins-base.  These packages are available in Ubuntu Lucid and Precise, and Debian Wheezy and Jessie, but not in Debian Stretch.

Stretch has python-gst1.0 and gstreamer1.0-plugins-base instead, but we don't know if they work with Gscreen, or what changes might be needed.

For now we've simply removed (c98aa61970639b5c02b3f255f9398fd5fe9de3b5) the dependency on python-gst0.10 and gstreamer0.10-plugins-base on Stretch, but this probably breaks that part of gscreen.  

Someone needs to make the appropriate changes, if any, that gscreen needs to support both gstreamer 0.10 and gstreamer 1.0, then we can add the dependency on the new packages to our Stretch packages.

SebKuzminsky, 2017-06-29:
See also #297.

jepler, 2017-07-18:
Fixed by fe2483c

closed 2017-07-18 by jepler