LinuxCNC/linuxcnc #300: Motion: Inconsistent axes limits

pkmcnc, 2017-07-06:
## Here are the steps I follow to reproduce the issue:

 1. Edit axis_mm to XZA configuration. Change
```
[TRAJ]
COORDINATES = X Z A

[KINS]
KINEMATICS = trivkins coordinates=xza

[AXIS_Z]
MIN_LIMIT = 0.0
MAX_LIMIT = 300.0

[JOINT_1]
MIN_LIMIT = -10.0
MAX_LIMIT = 310.0
HOME = 300.000
HOME_OFFSET = 310.0

[AXIS_A]
MIN_LIMIT = -999999.0
MAX_LIMIT = 999999.0

[JOINT_2]
TYPE = ANGULAR
MIN_LIMIT = -999999.0
MAX_LIMIT = 999999.0
HOME = 0.000
```
 2. Start the machine, home.
 3. In MDI tab execute any A move, like G1 A10 F300.

## This is what I expected to happen:
A moves

## This is what happened instead:
Error: **Linear move on line 0 would exceed Z's positive limit**

## Information about my hardware and software:

* I am using Debian image.
* I am running a binary I built myself from source.
* I am using this LinuxCNC version v2.8.0-pre1-3217-g367b5f8

pkmcnc, 2017-07-06:
X move also causes the same error.
But the problem seems to disappear after executing any Z move.

jepler, 2017-07-06:
If you increase the limit so that the home position is strictly inside it, e.g., `[AXIS_Z]MAX_LIMIT=300.01`, does the problem persist?  If that's the case, I think this is just a misconfiguration and maybe an opportunity to improve the documentation.  If not, then it's worth looking deeper for what the real problem is.

pkmcnc, 2017-07-06:
@jepler You are correct! Increasing the Z limit helped (but not joint 1 limit alone).
Interesting that if I move Z to say 200 and back to 300, everything works. The problem appears only until any Z move is executed.
Also, Z limit at 300.1 with joint 1 limit at 300 work OK.

closed 2017-07-06 by pkmcnc