LinuxCNC/linuxcnc #314: spindle stop-resume in G96 CSS mode

fixerslo, 2017-08-05:
If you hit halui.spindle.stop and halui.spindle.start in auto - paused the spindle mode is reset to G97 and the value of cutting speed is applied to spindle speed. 

## Here are the steps I follow to reproduce the issue:

Run that program, hit pause in G1 block and stop/restart spindle:
```
G7 G18 G54 G95 
G96 S170 D2000 M3

G0 X50 Z0
G1 Z-100 F0.1
M30
```

## This is what I expected to happen:
Spindle mode and speed reset to CSS mode with proper S value

## This is what happened instead:
spindle restarts in constant RPM mode, S value from G96 block used

massimodamassa, 2017-08-15:
Just put the modal commands in the .ini file

[RS274NGC]
PARAMETER_FILE = linuxcnc.var
RS274NGC_STARTUP_CODE = G7 G18 G21 G40 G54 G61 G80 G90 G95

fixerslo, 2017-08-15:
I guess I was not clear enough - english is not my first language, sorry. On my machine, I have external buttons to control spindle. They are connected to halui spindle pins. If I pause a program and stop spindle, for example to check on tool or clear chips, after restarting spindle and resuming program, the spindle speed is 170 rpm, instead of 170m/min (G96 S170 M3).
It is the same if I have G95 or G96 in INI startup codes.

massimodamassa, 2017-08-15:
The spindle like the power off, the external button what does M5 do?
In my so            'net spindle-manual-stop halui.spindle.stop
But I did not see how it behaves
Tomorrow I try and let you know.

I'm Italian and you?

massimodamassa, 2017-08-16:
I have verified, even to me the problem arises, though putting G97 in .ini
This is very dangerous.
We hope to be settled

robharp, 2017-09-14:
i have notcied the same also in tool touch off if you have been in  CSS mode. 

set spindle in CSS mode from MDI/Auto,
stop spindle M05

now in manual mode set spindle running manuly (fwd or rev)
touch off tool in X
spindle will increase to CSS speed on new length sometimes this leads to a full speed of spindle output.
sometimes i have seen the spindle reverse back to the direction it was in from Auto mode

pippin88, 2017-10-20:
Using Master and Gmoccapy:

Using G96, in any situation that you stop the spindle, G96 no longer works / applies.
Active GCodes still shows G96, however machine behaves like G97 (RPM mode) with S equal to the G96 S value.

E.g.: (using MDI)
M3 (start spindle)
G96 D2500 S120 (CSS works as expected)
Stop the spindle (with GUI button or M5)
Jog machine around or change tool
M3 - now spindle just turns at 120rpm and CSS not active, despite G96 showing as active G code.

Have to reapply G96 D2500 S120 after M3 every time.
It seems that you cannot apply G96 before the spindle is turning - but no error is shown if you try.

### Expected behaviour
G96 applies until G97 called, whatever happens to spindle M3/M4/M5

### Alternate acceptable behaviour
Machine defaults to G97 whenever M5 invoked, however there should be NO RPM / S value set, and therefore an error will show / active input is required from the operator

Attila1000, 2018-11-20:
Hello, 

i find this problem too. My HAL file call halui.spindle.stop if pressed the Spindle Stop button during program paused, and restarting spindle with halui.spindle.start, and the same situation happened (spindle starts with lowered speed in G94 mode). 
But if i use motion.spindle-inhibit instead of halui.spindle.start/stop the spindle stops, and restarts with desired speed. But the inhibit pin must driven by flipflop component.

fixerslo, 2019-07-16:
I found an (ugly?) fix for this. Since I don't use git, I am posting the solution here (I am using 2.7.14, in master it is somehow different due to multispindle).

emc_nml.hh:
```
class EMC_SPINDLE_STAT:public EMC_SPINDLE_STAT_MSG {
  public:
    EMC_SPINDLE_STAT();

    // For internal NML/CMS use only.
    void update(CMS * cms);

    double speed;		// spindle speed in RPMs
    double css_maximum; 	//<------------------  added for css bug fix
    int direction;		// 0 stopped, 1 forward, -1 reverse
    int brake;			// 0 released, 1 engaged
    int increasing;		// 1 increasing, -1 decreasing, 0 neither
    int enabled;		// non-zero means enabled
    int orient_state;
    int orient_fault;
};
```

emccanon.cc:
```
void SET_SPINDLE_MODE(double css_max) {
    css_maximum = fabs(css_max);
    emcStatus->motion.spindle.css_maximum = css_maximum; //<------------------  added for css bug fix
}
```

and finally halui.cc:
```

static int sendSpindleForward()
{
    EMC_SPINDLE_ON emc_spindle_on_msg;
    if (emcStatus->task.activeSettings[2] != 0) {
        if(emcStatus->motion.spindle.css_maximum){
		emc_spindle_on_msg.speed = fabs(emcStatus->motion.spindle.css_maximum); //fabs(emcStatus->task.activeSettings[2]);
		emc_spindle_on_msg.factor = 1000 / (2 * M_PI) * fabs(emcStatus->task.activeSettings[2]);
		emc_spindle_on_msg.xoffset = emcStatus->task.g5x_offset.tran.x + emcStatus->task.g92_offset.tran.x + emcStatus->task.toolOffset.tran.x;
	}
	else{
		emc_spindle_on_msg.speed = fabs(emcStatus->task.activeSettings[2]);
		emc_spindle_on_msg.factor = 0;
	}
    } 
    else {
	emc_spindle_on_msg.speed = +1;
	emc_spindle_on_msg.factor = 0;
    }
    return emcCommandSend(emc_spindle_on_msg);
}
```

besriworld, 2025-05-08:
It seems this problem still exists. I have the same case with linuxcnc 2.9.4