LinuxCNC/linuxcnc #315: [stepconf] charge-pump.enable pin not connected to the correct signal if an external e-stop switch is used.

thuSkywalker, 2017-08-06:
## Here are the steps I follow to reproduce the issue:

 1. Generate a machine config with the stepconf file below:
https://github.com/thuSkywalker/syil-x5-linuxcnc-config/blob/dd45ce7ea70a89a7dca43d7cb972000c5a38544f/Syil_X5.stepconf
Note that an input of parport is configured as e-stop function.
 2. The generated machine HAL file should be similar to this:
https://github.com/thuSkywalker/syil-x5-linuxcnc-config/blob/dd45ce7ea70a89a7dca43d7cb972000c5a38544f/Syil_X5/Syil_X5.hal
 3. Start LinuxCNC, turn on the Machine Power in axis GUI, attempt to jog the machine.

## This is what I expected to happen:
The charge pump is enabled, The machine moves.

## This is what happened instead:
The machine doesn't move because the charge pump isn't enabled.
By looking into the Syil_X5.hal file (link above) I found the enable pin of charge-pump is still connected to the estop-out signal (line 10), the same as when I didn't configure an e-stop input in stepconf. It should have been connected to estop-ext as I'm using an external e-stop switch and I did configure an e-stop input in stepconf.

andypugh, 2017-08-06:
I tried the config created by the stepconf file, and I agree that the behaviour seems unexpected. 
However, clicking the E-stop release (X) button in Axis does appear to turn on the charge pump. 
(the unexpected behaviour is that external e-stop activation on pin 11 does not turn the charge-pump off.

thuSkywalker, 2017-08-07:
@andypugh In my Axis 2.7.11 the e-stop button is always not pressed (with the configuration above), and the chare-pump.enable pin is always 0 no matter what I do.

andypugh, 2017-08-07:
Have you tried pressing the button? (on-screen?).  What is the value of your pin 11 input?

thuSkywalker, 2017-08-13:
@andypugh Yeah, sure, tried everything I could. My pin 11 input is high in the normal state (non-estop), low when the e-stop button on my machine is pressed.