LinuxCNC/linuxcnc #316: Request new stepconf feature

CodePuse, 2017-08-10:
Addition of de-bounce check box (and underlying code) to limit and home switch configuration.
Thanks.

nicokid, 2017-08-24:
Can you please explain it better?

Nicola.

nicokid, 2017-08-24:
Sorry, I did not read the previous thread.
Wait for me to read it.
Nicola.

CodePuse, 2017-08-24:
I found that I had a spontaneous limit switch trigger while routing ally. There were various solutions proposed in fora including using debounce in the hal file. 

I thought it would be useful/convenient to add debounce and the time period to the stepconf. On the screen where you set the pin (on the parallel interface board) to its function ie limit/home etc, add an additional check box (debounce or not) and text box (time period). These would add the appropriate code to the hal file when the stepconf wizard is finished.

nicokid, 2017-08-24:
Adding a text box for each pin takes up so much space.
Is it appropriate to put a single text box with a global delay?

Nicola.

CodePuse, 2017-08-25:
Honestly what ever you think would work best. I just wondered whether debounce could be added to stepconf rather than had to hand code the hal file. 

I assume you don't need a different delay value for each pin. I have used the same delay value myself. Are you actually adding this function or say evaluating it for addition? Thanks.

nicokid, 2017-08-26:
I gave a first look at the debounce, and it should not be a big deal.
However, I have to wait for Jeff Epler to approve my last modifications before changing other things.

Nicola.

c-morley, 2017-08-26:
One way would be to change the current inverse checkbox to a combobox and adding both options to the combobox.

Chris M

----- Reply message -----

nicokid, 2017-08-28:
Uhmm, I'm not sure about this solution. The combobox should have this 4 entries:
invert no debounce
invert + debounce
no debounce
debounce
No so pretty...

Nicola.

jepler, 2017-08-28:
I think if you are going to implement debounce in stepconf, you should simply debounce every input that is "expected to only have slow transitions", either unconditionally or controlled by a single checkbox.  That would be everything but spindle A/B/Index.

Based on the fact that products such as https://www.maximintegrated.com/en/products/interface/signal-line-protection-ics/switch-debouncers-protectors/MAX6817.html use a pretty large debounce time of 40ms nominal, you should probably choose a debounce time within that range.

stepconf currently chooses a HOME_LATCH_VEL like so:
~~~~
            # set latch velocity to one step every two servo periods
            # to ensure that we can capture the position to within one step
            latchvel = latchvel * 500 / get("scale")
            # don't do the latch move faster than the search move
            if abs(latchvel) > abs(get("homevel")):
                latchvel = latchvel * (abs(get("homevel"))/abs(latchvel))
~~~~
but of course this math is bunk when there's a 40ms debounce time.  With a 40-servo-period debounce, the offset becomes about 20 steps using this math.  With such aggressive debounce, though, you may not be able to reach the goal of homing to within a single step.  Anyway, it should be a pretty constant offset, and it looks like for reasonable speed and step size settings it's well into the sub-mm range.

c-morley, 2017-08-28:
Sorry I was imagining adding two check boxes to the combobox.
You can't do it in glade editor but can with python programming.

Just an idea
Chris M

----- Reply message -----

nicokid, 2017-08-28:
Hi Chris,
I was just trying to put check boxes into a combobox. However, the options would no longer be immediately visible and we should select all the combo boxes to check the single settings. Not a good deal for the user.

Nicola.

c-morley, 2017-08-28:
That's too bad..I agree must be reasonable for users to find.

Chris M

----- Reply message -----

c-morley, 2017-08-28:
I think it's possible to add icons to combboxes also. If one used small icons they may be visible.
Another idea is a button that opens a dialog that allows selection of individual options.

Chris M

----- Reply message -----

nicokid, 2017-08-29:
I think that in the end I will just add another check box. But it seems to me that Jeff Epler does not like debounce.

Nicola.

jepler, 2017-08-30:
Debounce is fine with me @nicokid

andypugh, 2017-08-30:
On my parallel-port machine I found that a 15-base-cycle debounce was entirely adequate. I don't think that there is any reason to push it out as far as 40mS. 
40mS sounds like something targeting mains-hum, not the type of high-frequency transient that we see in our machines. 
Potentially stepconf could debounce the home switches for 10 cycles and the the limit switches for 50 cycles (or more) as a noisy home is a lot less trouble than a noisy limit. 

I suggest two check-boxes at the bottom of the port-setup page with a default time pre-filled:
[X] Debounce Home switches [0.2mS]
[X] Debounce Limit switches [1.0 mS]

jepler, 2017-08-30:
switch bounce over 8ms: http://www.ganssle.com/debouncing.htm
switch bounce over 40ms: https://softsolder.com/2012/07/13/contact-bounce-why-capacitors-dont-fix-it/

Since limit and home switches are only sampled by the servo thread, a debounce of <1ms doesn't really make a difference anyway, so users requesting to add debounce have switches that bounce for >1ms. (A/B/Z of spindle are a different story, obviously)

jepler, 2017-08-30:
> As shown in the figure at the right, the contact switching time is the sum of the unstable time, reverse time, and bounce time, which is 5 to 15 ms for general Basic Switches
https://www.ia.omron.com/support/guide/29/explanation_of_terms.html

andypugh, 2017-08-30:
Jepler: I think that there is some confusion here. The "debounce" component is not typically used to ignore physical switch bounce, it is used to reject transient electrical noise in the limit switch signals.
With a stepconf system the inputs are sampled every base-thread, and a base-thread debounce component can be used to pass on a filtered switch state to the servo-thread motion inputs. 
Thus with a parallel port system it _can_ make sense to have a shorter-than-servo-period debounce time.

jepler, 2017-08-31:
@andypugh thanks for the clarification. If there a clearer term to show to users? Of perhaps it was clear to everyone but me.

andypugh, 2017-08-31:
It might make sense for the Stepconf GUI to have 
[X] Use software filtering on Limit inputs
[X] Use software filtering in Home inputs