LinuxCNC/linuxcnc #321: axis: hangs opening an incorrect file

l29ah, 2017-09-02:
## Here are the steps I follow to reproduce the issue:

 1. create a g-code file starting with `"`
 2. open it with axis

## This is what I expected to happen:
axis complaining about incorrect file contents

## This is what happened instead:
axis stops responding to any user input

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): Gentoo
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [x] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): v2.8.0-pre1-3315-ga7aeaa608
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): parport

jepler, 2017-09-03:
@l29ah thank you for your report.

Can you provide more information on how to reproduce the problem?  I tested with linuxcnc master branch and `configs/sim/axis/axis.ini`.  First, I created a file called `doublequote.ngc` with the content
~~~~
" issue 321
~~~~
when I opened it in axis with File > Open, I got a pop-up which stated "Near line 2 of /home/jepler/doublequote.ngc: Bad character '"' used".  After dismissing this pop-up, AXIS and LinuxCNC continued as usual.

I also tried renaming the file to `".ngc` in case you meant that the problem related to the name of the file instead of the content of the file.  However, this also didn't produce a problem.

I tested at `v2.8.0-pre1-3316-g9346a55a3`.  The difference from -3315 to -3316 is not likely to be relevant.

jethornton, 2017-09-03:
Could this be an issue with a file saved with a windows editor or a word processor?

l29ah, 2017-09-03:
https://dump.bitcheese.net/files/wicibaj/table-zig.ngc there's my file

jepler, 2017-09-03:
@jethornton the file @l29ah attached has UNIX line endings, so that's not it.

Unfortunately, I don't reproduce the problem using table-zig.ngc either.  Does it reproduce for you when using the sample configuration `configs/sim/axis/axis.ini`?

jethornton, 2017-09-03:
@jepler , I get the same result as you using the Axis sim, Bad character '"' used.

l29ah, 2017-09-03:
Just reproduced it with a file containing `"test`. Maybe it reproduces only when linuxcnc is ran over ssh's X forwarding facility; other pop-ups like the file selection dialog work well tho.

jepler, 2017-09-03:
@l29ah what if it's another invalid program not involving the `"` character?  For instance, if the first and only line is `G999` then you should get the message "G code out of range", created by the same code that would show the pop-up in the other case.

jepler, 2017-09-03:
@l29ah are you using a different user interface language besides English?

l29ah, 2017-09-03:
Yes, with `G999` it's the same. No, English here.