LinuxCNC/linuxcnc #325: missing runtime dependency check in src/configure

SebKuzminsky, 2017-09-08:
Fresh install of Xubuntu 17.04 "Zesty", checked out the master branch, ran "debian/configure uspace", satisfied dpkg-checkbuilddeps.  Built linuxcnc and ran it, selected the sim/axis/axis config, and got this error:

~~~
Traceback (most recent call last):
  File "/home/seb/linuxcnc-dev/bin/axis", line 3784, in <module>
    get_coordinate_font(vars.dro_large_font.get())
  File "/home/seb/linuxcnc-dev/bin/axis", line 3680, in get_coordinate_font
    glnav.use_pango_font(coordinate_font, 0, 128)
  File "/home/seb/linuxcnc-dev/lib/python/glnav.py", line 6, in use_pango_font
    import pango, cairo, pangocairo
ImportError: No module named pango
Shutting down and cleaning up LinuxCNC...
~~~

Looks like a missing runtime dependency check in src/configure.  Fix should go in 2.7.

SebKuzminsky, 2017-09-08:
Installing python-gtk2 fixed it.

jepler, 2017-09-08:
It's probably not a bug in the debian packaging ,but maybe a bug in how we think about setting up new systems to run LinuxCNC-built-from-source.

jepler, 2017-09-08:
(that is, it's totally not a bug for a source package to *NOT* depend on something that's not used until runtime)

SebKuzminsky, 2017-09-08:
I agree.

Just like how `src/configure` without `--disable-check-runtime-deps` reminds the user to install bwidget, libtk-img, and tclx, since they are needed to make use of a run-in-place build, so it should remind the user to install python-gtk2 (or Axis won't run).

commit 2f6eafb4 (2017-09-10): src/configure: make sure python's pango and cairo modules are installed

These modules are needed to run Axis, and possibly other things.
This test can be skipped with --disable-check-runtime-deps, like the
rest of the runtime dependency checks.

Fixes #325.
closed 2017-09-10 by SebKuzminsky