LinuxCNC/linuxcnc #329: emclcd: invokes deprecated commands to find network info

jepler, 2017-09-14:
I was reading at https://dougvitale.wordpress.com/2011/12/21/deprecated-linux-networking-commands-and-their-replacements/ and it made me grep around in LinuxCNC to see if we had any instances of these (I worried I might have written them in hm2-eth).  It turns out that emclcd is invoking "ifconfig" and "netstat".  These will stop working on new Linux installs.

jepler, 2017-09-14:
More concretely, this affects Debian Stretch which does not install ifconfig or route.

andypugh, 2017-09-14:
As far as I know emclcd has never actually worked, and only supports a particular LCD interface which is currently too hard to buy. 
Despite that, I think we should keep it in our tree but consider not compiling it. I like it as an exemplar of a very different UI concept.