LinuxCNC/linuxcnc #335: linuxcnc master build failed

midyukov-anton, 2017-10-02:
I'm trying to build linuxcnc from the master branch. I'm getting the error:
```
+ exec time rpmbuild --rebuild --target=x86_64 linuxcnc-2.8.0-alt1.290917.1.src.rpm
Installing linuxcnc-2.8.0-alt1.290917.1.src.rpm
Building target platforms: x86_64
Building for target x86_64
Executing(%prep): /bin/sh -e /usr/src/tmp/rpm-tmp.83514
+ umask 022
+ /bin/mkdir -p /usr/src/RPM/BUILD
+ cd /usr/src/RPM/BUILD
+ cd /usr/src/RPM/BUILD
+ rm -rf linuxcnc-2.8.0
+ echo 'Source #0 (linuxcnc-2.8.0.tar):'
Source #0 (linuxcnc-2.8.0.tar):
+ /bin/tar -xf /usr/src/RPM/SOURCES/linuxcnc-2.8.0.tar
+ cd linuxcnc-2.8.0
+ /bin/chmod -c -Rf u+rwX,go-w .
+ sed 's/ -o root//g' -i src/Makefile
+ exit 0
Executing(%build): /bin/sh -e /usr/src/tmp/rpm-tmp.17990
+ umask 022
+ /bin/mkdir -p /usr/src/RPM/BUILD
+ cd /usr/src/RPM/BUILD
+ cd linuxcnc-2.8.0
+ pushd src
~/RPM/BUILD/linuxcnc-2.8.0/src ~/RPM/BUILD/linuxcnc-2.8.0
+ autoreconf -fisv
autoreconf-default: Entering directory `.'
autoreconf-default: configure.in: not using Gettext
autoreconf-default: running: aclocal --force 
aclocal-default: warning: autoconf input should be named 'configure.ac', not 'configure.in'
autoreconf-default: configure.in: tracing
autoreconf-default: configure.in: not using Libtool
autoreconf-default: configure.in: not using Intltool
autoreconf-default: configure.in: not using gtk-doc
autoreconf-default: running: /usr/bin/autoconf-2.60 --force
autoreconf-default: running: /usr/bin/autoheader-2.60 --force
autoreconf-default: configure.in: not using Automake
autoreconf-default: Leaving directory `.'
+ CFLAGS='-pipe -Wall -g -O2'
+ export CFLAGS
+ CXXFLAGS='-pipe -Wall -g -O2'
+ export CXXFLAGS
+ FFLAGS='-pipe -Wall -g -O2'
+ export FFLAGS
+ '[' -n '' ']'
++ sed -r 's/(^|[[:space:]]+)-[^m][^[:space:]]*//g'
++ printf %s '-pipe -Wall -g -O2'
+ ASFLAGS=
+ export ASFLAGS
+ export lt_cv_deplibs_check_method=pass_all
+ lt_cv_deplibs_check_method=pass_all
+ xargs -rn1 install -pm755 -- /usr/share/gnu-config/config.sub /usr/share/gnu-config/config.guess
+ sort -u
+ xargs -ri find '{}' -type f '(' -name config.sub -or -name config.guess ')' -printf '%h/\n'
+ xargs -ri dirname -- '{}'
+ readlink -e -- ./configure
+ ./configure --build=x86_64-alt-linux --host=x86_64-alt-linux --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --sysconfdir=/etc --datadir=/usr/share --includedir=/usr/include --libdir=/usr/lib64 --libexecdir=/usr/lib --localstatedir=/var/lib --sharedstatedir=/var/lib --mandir=/usr/share/man --infodir=/usr/share/info --disable-dependency-tracking --without-included-gettext --enable-non-distributable=yes --with-realtime=uspace
configure: WARNING: unrecognized options: --disable-dependency-tracking, --without-included-gettext
checking for x86_64-alt-linux-g++... x86_64-alt-linux-g++
checking whether the C++ compiler works... yes
checking for C++ compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C++ compiler... yes
checking whether x86_64-alt-linux-g++ accepts -g... yes
checking how to run the C++ preprocessor... x86_64-alt-linux-g++ -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
[... 1460 lines cut ...]
Copying python script notification.py
Syntax checking python script preferences.py
Copying python script player.py
Syntax checking python script widgets.py
Copying python script preferences.py
Copying python script widgets.py
Copying glade file gmoccapy.glade
Compiling emc/kinematics/lineardeltakins.cc
Compiling emc/kinematics/rotarydeltakins.cc
/usr/include/boost/math/special_functions/fpclassify.hpp: In function 'bool boost::math_detail::is_nan_helper(__float128, const true_type&)':
emc/kinematics/lineardeltakins.cc:18:18: error: 'std::std' has not been declared
 #define isnan(x) std::isnan(x)
                  ^
/usr/include/boost/math/special_functions/fpclassify.hpp: In function 'bool boost::math_detail::is_nan_helper(__float128, const false_type&)':
emc/kinematics/lineardeltakins.cc:18:18: error: 'std::std' has not been declared
 #define isnan(x) std::isnan(x)
                  ^
Compiling emc/rs274ngc/gcodemodule.cc
make: *** [objects/emc/kinematics/lineardeltakins.o] Error 1
make: *** Waiting for unfinished jobs....
make: Leaving directory `/usr/src/RPM/BUILD/linuxcnc-2.8.0/src'
error: Bad exit status from /usr/src/tmp/rpm-tmp.17990 (%build)
```

Build 2.7.11 is successful. Distribution ALT Linux Sisyphus, gcc 6, boost 1.65

jepler, 2017-10-02:
In `src/emc/kinematics/lineardeltakins.cc`, I recommend trying one of the following:
- change the `#define isnan` line to `using std::isnan`
- reorder the lines like so:
 ````
#include <boost/python.hpp>
using namespace boost::python;
#define isnan(x) std::isnan(x)
#include "lineardeltakins-common.h"
````
I tested both alternatives on Debian Stretch, and they compile there (but of course the original did too)

midyukov-anton, 2017-10-03:
>     reorder the lines like so:
> 
> #include <boost/python.hpp>
> using namespace boost::python;
> #define isnan(x) std::isnan(x)
> #include "lineardeltakins-common.h"

Now build is successful. Thanks.

jepler, 2017-10-03:
OK, I look forward to your pull request so we can get it fixed here as well.

midyukov-anton, 2017-10-04:
Created pull request #338

jepler, 2017-10-14:
Closed via #338

closed 2017-10-14 by jepler