LinuxCNC/linuxcnc #341: halgui: uses python-gnome2, which will be removed in buster

jepler, 2017-10-20:
Reportedly, many "obsolete" gnome packages, including python-gnome2, will be removed in Debian Buster.  This appears to affect halgui.

SebKuzminsky, 2017-10-20:
halgui was written in 2005 by Daniel Nilsson and never touched after that.  It was not integrated into our build system, so it's not included in our packages.  The only mention in the docs is this:

~~~~~
halgui::
    GUI tool for configuration and tuning (not implemented yet).
~~~~~

I propose we remove it in master.

jepler, 2017-10-20:
I may be mistaken that halgui is the only part that depends on python-gnome2.  It appears the original addition of python-gnome2 as a dependency was for stepconf: f37e7389f027c0b843dcf88aed59eb3b96e55ce4

possibly relevant, the related `.glade` file says `<requires lib="gnome"/>`, which is not in stepconf anymore in master branch, but is in pncconf.  I think stepconf has been converted to glade3, while pncconf remains on gnome2.

Anyway, I do not oppose the removal of halgui, particularly if @SebKuzminsky is right that it's never been usable in LinuxCNC packages since its addition 12 years ago (oh my)

jepler, 2017-10-20:
.. after removing package `python-gnome2` on my stretch machine, pncconf still starts and seems to work.  Tested on recent master branch.  So maybe <requires lib="gnome"> is pulling in something besides python-gnome2, or maybe it's redundant and should be removed.  It does access this file,
~~~~
stat("/usr/lib/x86_64-linux-gnu/gio/modules/libgiognomeproxy.so", {st_mode=S_IFREG|0644, st_size=18896, ...}) = 0
~~~~
from package glib-networking, for whatever that's worth.

c-morley, 2017-12-04:
pncconf and stepconf don't require gnome packages anymore. It was one of the reasons that they needed to be changed to use tab based widget rather then the gnome wizard widget.
Any reference to gnome _should_ only be incidental.

closed 2024-10-29 by jepler