LinuxCNC/linuxcnc #361: axis returns linuxcnc to MODE_MANUAL after each MDI command

KurtJacobson, 2017-11-30:
## Here are the steps I follow to reproduce the issue:

 1. Start LinuxCNC using sim axis.ini
 2. Power on and home the machine
 3. Open a terminal and if needed set up the RIP environment
 4. Launch an interactive python console and run the following commands
```
>>> import linuxcnc
>>> s = linuxcnc.stat()
>>> c = linuxcnc.command()
>>> s.poll(); print s.task_mode
1
>>> c.mode(linuxcnc.MODE_MDI)
>>> s.poll(); print s.task_mode
3
>>> c.mdi('G0 X1')
>>> s.poll(); print s.task_mode
1 # Back to MANUAL_MODE, subsequent MDI commands will fail 
>>> c.mdi('G0 X0') # ERROR: "Must be in MDI mode to issue MDI command"
>>> 
```

## This is what I expected to happen:

I expect LinuxCNC to remain in MDI mode until the mode is changed via `c.mode()`, or by some interaction with the UI.

## This is what happened instead:

It appears that axis is setting LinuxCNC to manual mode whenever the interpreter becomes idle, which causes subsequent MDI commands to fail.  

## It worked properly before this:

Works as expected in 2.7.11,  i.e. LinuxCNC remains in MDI mode so that multiple MDI commands can be issued in succession without having to set MDI mode before each command. 

## Information about my hardware and software:

 * I am running: A binary I built myself
 * I am using this LinuxCNC version:  v2.8.0-pre1-3395-gbd25a5d11
 * I am using this user interface (GUI): axis 

[1]: https://github.com/verser-git/probe_screen

KurtJacobson, 2017-11-30:
I have been unable to find the commit that introduced this change, but I remember something about making axis return LinuxCNC to manual mode after an MDI. I believe this was to resolve the nuisance in 2.7 of having to first jog via the keyboard to set LinuxCNC to manual mode before a MPG could be used to jog.

Returning to manual mode seems like desirable behavior in general. The problem is that it has broken at least one user plugin (right term?) [probe_screen][1], which issues a series of MDI commands via the python interface and expects LinuxCNC to remain in MDI mode after each command has been executed.

I have tried setting linuxcnc to MDI mode immediately before issuing each MDI command but that fails maybe 50% of the time. I guess axis is setting the mode back to manual between when the mode is set to MDI and when the MDI command is actualy issued.

[1]: https://github.com/verser-git/probe_screen

KurtJacobson, 2017-11-30:
This crude script demonstrates that even if the mode it set to MDI immediately before issuing an MDI command, somehow either LinuxnCNC does not actually get set to MDI, or it gets set back to to manual between the time the scripts sets the mode to MDI and actual issues the MDI command.

You might have to run this a couple times, but it should cause the error `Must be in MDI mode to issue MDI command` to raise its ugly head.

I hope I am just doing something stupid here, and there is an easy solution.

```python
#!/usr/bin/env python

import linuxcnc

s = linuxcnc.stat()
c = linuxcnc.command()

MODES = {
    linuxcnc.MODE_MANUAL: 'MANUAL',
    linuxcnc.MODE_AUTO  : 'AUTO',
    linuxcnc.MODE_MDI   : 'MDI'
    }

def print_mode():
    s.poll()
    print "Current MODE: ", MODES[s.task_mode]

print_mode()

for i in range(4):
    print '\n'
    print_mode()

    print 'Changing to MDI mode ...'
    c.mode(linuxcnc.MODE_MDI)
    c.wait_complete()

    print_mode()

    cmd = 'G0 X{}'.format(i)
    print 'Issuing MDI command: ', cmd
    c.mdi(cmd)
    c.wait_complete()
```

gmoccapy, 2017-11-30:
As this happen also with gmoccapy, i think it is not related to the gui itself.Norbert

KurtJacobson, 2017-11-30:
That is interesting Norbert, I had not noticed it happening in Gmoccapy, I will test it again.

I do think it is GUI dependent though as my little script behaves correctly when I run it using hazzy as the GUI.

gmoccapy, 2017-11-30:
I had no possibility to test with your script, but probe-screen embedded in gmoccapy does also complain about not being in MDI mods.Norbert

KurtJacobson, 2017-11-30:
OK, I made two quick videos showing the difference between running the scrip I posted above in axis and in hazzy.  When run in hazzy it behaves as I would expect, stopping between each 1in segment due to the `c.wait_complete()`. Also the `Must be in MDI mode to issue MDI command` error does not occur. Note that the interp state displaced in the 'BasicControls'  widget briefly changes from READ to IDLE between each segment, but the mode remains in MDI.

When I run the script in axis, the first segment seems to run correctly, but the second two segments are 'run together'. You can see this both from the rapid succession of prints in the terminal and the motion of the cone in the back-plot. Also, we get the `Must be in MDI mode to issue MDI command` error. Also if you look at the LinuxCNC Status display you will see that the task_mode in jumping around between mdi and manual.

I think this indicates that the problem is with the GUI, and not elsewhere.

Here are links to the two videos:
hazzy:  https://youtu.be/1F0vA37Maew
axis:  https://youtu.be/BFbfgQmI7gY

(Sorry for testing against hazzy, since it is not a standard UI, but I know how it works (I hope :D) so that was the easiest way for me to eliminate other possible causes.)

pkmcnc, 2017-11-30:
I encountered this problem too, see #285

KurtJacobson, 2017-12-01:
Thanks @pkmcnc, I saw your issue but I was not certain it was the same thing, now I am pretty sure it is. Sorry for the duplication.

pkmcnc, 2017-12-01:
My description was not clear enough, probably. I'm kind of glad that you also found this issue. It's pretty annoying, because I have a blocking relay connected to manual mode pin (inverted, actually), and it goes on/off when the machine starts, when MDI tab is selected, also before and after every MDI command.

KurtJacobson, 2017-12-01:
@gmoccapy I don't know how good of a test my little script is, but I just tested Gmoccapy with it and everything seems to work as expected. When I get a chance I will test Gmoccay with probe_screen.

rene-dev, 2017-12-01:
Good news: I think I found the commit that breaks it.
Bad news: its this one: https://github.com/LinuxCNC/linuxcnc/commit/a2f0de5
I tried to test which of the JA commits breaks this, but I cant get any of them to build.
I also noticed, that when you click on manual or mdi in axis, task_mode is not updated properly on all versions after this commit, and I think this is the same issue as #285
I think also breaks @andypugh s lathe macros: https://forum.linuxcnc.org/41-guis/26550-lathe-macros

rene-dev, 2017-12-01:
Hmm, maybe this is an axis bug?
the mode switching and test script work ok on gmoccapy.
havent tested the probe screen...

rene-dev, 2017-12-02:
fix from @dngarrett https://github.com/LinuxCNC/linuxcnc/commit/3de8778374bcee0b184c0c4ad04a373585efb7e0

rene-dev, 2017-12-13:
I think this can be closed.

jepler, 2017-12-13:
@rene-dev closing per your note. Thanks @dngarrett

closed 2017-12-13 by jepler

hnyangzl, 2017-12-28:
I have Meet the same problem，How can I  solve this problem?  thank you