LinuxCNC/linuxcnc #371: Feature suggest: Add an "is-probing" pin to motmod?

ReeseWang, 2017-12-04:
I'm writing some auto measurement G-code for my milling machine with a touch probe.
I definitely don't want to crash the probe.
But LinuxCNC (2.7.11) seems don't want to stop when the probe is tripped during a non-probe move if the machine is running G-code program (AUTO mode IIRC).

SebKuzminsky, 2017-12-04:
I think the goal of having accidental/unexpected probe trips abort any motion is a good idea.

Does @dngarrett 's recent fix in master fix this for you?  It's commit e10c52ea, available in v2.8.0-pre1-3405-ge10c52e and later.

If not, can you add a note to this issue, or to #368 and/or #369?

Also note that motion has a pin named motion.motion-type, which is set to 5 only when in a probing move: http://linuxcnc.org/docs/devel/html/man/man9/motion.9.html#MOTION PINS

rene-dev, 2017-12-04:
the other issues only address stopping during homing/jogging/mdi, but not auto mode.
I just checked, and it does not care about the probe during auto mode.
stopping during any non-probe move anytime sounds good.

Lcvette, 2019-10-11:
I agree, i am on version 2.9 but also tried with 2.8, running debian 9 the stretch test iso, i have written a entire probing routine library for the new probe_basic gui but have been getting reports along with my own testing of the probe trigger during non probe moves not caring and causing crashes. I am assuming nothing has been fixed yet?  if so is there a way to fix it or a work around?  maybe something added to the ngc probe routine file?

thanks in advance!

Chris