LinuxCNC/linuxcnc #372: Remaping existing gcodes does not work (example G74); free gcodes work

emcPT, 2017-12-05:
2.8.0 pre

Remapping G74 (possible others) does not work - linuxcnc still try to run the **original built in G74 code**. Creating a new code, like G72 does work, with the remapped code . Was functional in 2.7.x as the python code was copy pasted to the new version.

This was detected on a lathe. When I try to run the remapped G74, linuxcnc complains that I cannot run G74 canned cycle in a machine without a Y axis (that is true for the standard G74).

andypugh, 2017-12-05:
The docs for 2.7 state that only a very few existing codes can be remapped. 
http://linuxcnc.org/docs/2.7/html/remap/remap.html#remap:remappable-codes

Are you sure that the remap worked before?

emcPT, 2017-12-06:
Issue should now be closed.
Explanation: On 2.7.x there was no implementation of any G74 and in 2.8.0 the G74 is already assigned to "floating tap left hand cycle".
I never realized this and I copy paste the working remap code from one version to another. This explains why it was working in 2.7.x and not in the 2.8.x version.

closed 2017-12-06 by emcPT