LinuxCNC/linuxcnc #374: G96 problem in 2.8.0 pre

emcPT, 2017-12-05:
This is probably related with issue #314 

Running 2.8.0, lathe machine

When a program is finished and the last spindle mode was G96 (constant surface speed), when loading any other program (including the exactly same program that was just completed), the error "S word missing with G96" shows up, even if the program is ok. My turn arround solution was to issue in MDI "G97", then I can load any gcode file that contains G96 again.

This behaviour was not present in 2.7

andypugh, 2018-10-04:
Did this problem appear in the last week? 
(If so, it might be a problem with the multi-spindle patch)

rene-dev, 2018-10-04:
it did not. I would like to know why the behaviour is different in 2.7. this bug is very annoying. 
hints:
https://github.com/LinuxCNC/linuxcnc/search?q=spindle_mode&unscoped_q=spindle_mode

zultron, 2020-06-08:
Is this issue purposely a 2.8 milestone?  It does affect 2.8, but it doesn't seem like it affects community enough to generate additional information for reproducing, testing, fixing.  #314, referred to in the OP as a possible cause, has much more information but is NOT set as a milestone.

Let's remove this as a 2.8 milestone, or else add #314, either way.