LinuxCNC/linuxcnc #378: limit3 constraint violation

SebKuzminsky, 2017-12-11:
We've discovered two glitches in @zultron's new limit3 component.  A new test added in 3eb0a7a1b247aa34bb80549c7b045b56ba1007c6 reproduces both.

The first glitch is when limit3's input abruptly begins to track its output.  This bug was fixed in a0fb2781268ac1c5eec049bf7ea6b4fa2ebf84c3.

The second glitch is an accel constraint violation that seems related to hunting for a constant-velocity input.
[image]

zultron, 2017-12-11:
Refs to #351, #240.  Thanks, @SebKuzminsky.  I'll take a look.

One of these may have been fixed over the weekend by @dngarrett and @samcoinc (and maybe others).

zultron, 2017-12-12:
I have a fix for this, and will clean up and submit PR by tomorrow.

The fix is trivial; just a case of being sloppy with values persisting between calls.

zultron, 2017-12-14:
I've gone off in the weeds a bit on this one, but I'll submit a PR by tomorrow for real.  :P 

In addition to the present fix, I also fixed the math a bit, and the overshoot problem is greatly reduced.

zultron, 2018-01-12:
So much for submitting a PR "by tomorrow for real", 29 days later.  My apologies.

PR #396 should fix these issues.

samcoinc, 2018-01-12:
Can't rush into these things.  :) The violation is certainly gone - initial testing looks awesome!.

commit 2e741d5e (2017-12-17): limit3:  fix accel violations when 'locking'

When locking onto the input signal or a max/min limit, don't assume
the desired velocity (input velocity for input signal, 0 for limit)
will be legal the next period.

This fixes a one-period acceleration spike that violated accel limits.

A following patch will clean up the `SET_NEXT_STATE()` macro.

Fixes #378
PR #396 (2018-01-13): 2.7:  limit3--fix acceleration
These patches address the shortcomings of PR #380, and fix #378, reporting constraint violations in the `limit3` comp.  Take a look at the top commit log for the specifics of what's going on here.
closed 2018-01-14 by SebKuzminsky