LinuxCNC/linuxcnc #379: posemath library is incorrect

lethang12cdt, 2017-12-13:
Hi,
 * I am using this LinuxCNC version 2.8.1, there are some incorrect functions in _posemath.c. I made a topic about it in this link: 
https://forum.linuxcnc.org/38-general-linuxcnc-questions/33688-posemath-library-is-incorrect#103080

andypugh, 2017-12-16:
It seems that this issue has been mentioned before:
https://forum.linuxcnc.org/forum/30-cnc-machines/26392-the-study-of-arithmetic

Where would we expect to see problems if this really is wrong?

lethang12cdt, 2017-12-16:
> the-study-of-arithmetic

In my RPY rotation matrix that i mentioned, when pitch(P) = +- pi/2, we
cant compute roll and yall by fomular  rpy->r = atan2(m->y.z, m->z.z); rpy->y
= atan2(m->x.y, m->x.x);
They are always atan2(0,0)

jepler, 2017-12-16:
According to the I changed the body of the function in question, pmMatRpyConvert to just `abort()`.  The result of the testsuite: 
Runtest: 202 tests run, 202 successful, 0 failed + 0 expected

in other words, our normal testing doesn't even call this function.

It appears that the function is used indirectly by pumakins, though.  pumakins has no tests.

andypugh, 2017-12-16:
I have a feeling that pumakins is also often reported to be flaky.

rene-dev, 2017-12-16:
is pumakins not just a less generic version of genserkins?

andypugh, 2017-12-16:
I am not particularly adept at tracing includes, but it doesn't seem to me that genserkins uses posemath at all.

lethang12cdt, 2017-12-18:
i found out that genserkins  uses  gomath library, the "go_mat_rpy_convert"
function is similar  "pmMatRpyConvert"

andypugh, 2017-12-18:
gomath uses GO_PI_2 in the same conditional, but that is defined as PI/2 
https://github.com/LinuxCNC/linuxcnc/blob/af15a4d90e1d51d5309db65fe1c9511e486df411/src/libnml/posemath/gomath.h#L52

So at least one of them is probably wrong. 
(and has been for about 10 years)

rene-dev, 2017-12-18:
maybe the 2 can be merged together? or replaced by a dependency? http://glm.g-truc.net/0.9.8/api/modules.html

jepler, 2017-12-18:
No, we cannot depend on a C++ library such as glm here, because this code needs to be compiled to run in Linux kernel space (for some users and configurations), and without unmaintainable trickery we can't do that with C++ code.

Yes, the duplication in posemath and gomath is unfortunate, all the more so because both originated at NIST in different times; The ancestor of LinuxCNC itself originated at NIST as EMC and always used posemath.  gomath was added later when genserkins was added.  Neither posemath nor gomath really seem to be maintained.  I checked the last NIST posemath release and it has the exact same code as LinuxCNC does today for pmMatRpyConvert.  https://github.com/usnistgov/rcslib/blob/master/src/posemath/_posemath.c seems to be an official mirror based on the github username.

rene-dev, 2017-12-18:
why does it need to run in kernel space? everything can run in userspace with rt preempt.

jepler, 2017-12-18:
Because we support RTAI in a mode which uses kernel modules.  Yes, if someday we drop this mode, then we can revisit the decision about C++ in realtime components.

rene-dev, 2017-12-18:
maybe genserkins needs a test, would be interesting if it behaves differently with the change.

rene-dev, 2017-12-18:
is there anything that works with RTAI, but not rt preempt? In my experience rtai is troublesome with never kernels and 64 bit.

jepler, 2017-12-18:
[this is not the place for a referendum on RTAI]

lethang12cdt, 2017-12-18:
The function "pmMatZyxConvert" have same error like  "pmMatRpyConvert".
https://github.com/LinuxCNC/linuxcnc/blob/master/src/libnml/posemath/_posemath.c#L522

pkmcnc, 2017-12-18:
@lethang12cdt 
Using pmMatRpyConvert after pmRpyMatConvert should produce the original vector. If not, make the proposed change and test again.

jepler, 2017-12-20:
actually it appears I'm mistaken, upstream has the same fix that this bug is proposing.  I failed at reading the source.  (just taking a copy of upstream fails, as they've incompatibly changed their APIs in the last hundred years)

I still wish we had a test.

closed 2017-12-20 by jepler

jepler, 2017-12-20:
(Despite the affects-2.7 label, I have fixed this in master only)

lethang12cdt, 2017-12-25:
the gomath.c library has errors too:
https://github.com/LinuxCNC/linuxcnc/blob/master/src/libnml/posemath/gomath.c#L423
https://github.com/LinuxCNC/linuxcnc/blob/master/src/libnml/posemath/gomath.c#L443
replace "GO_PI_2" to "-GO_PI_2" for conditions.

there is an error in genserkin.c too, in main() function add value of
interation for KinematicInverse function, if not it's wont work when we
test in user mode
recommend: add
KINS_PTR->max_iterations = GENSER_DEFAULT_MAX_ITERATIONS;

 at:
https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/kinematics/genserkins.c#L724

jepler, 2017-12-25:
Please open pull requests for the problems you identified. Please open one pull request for each separate problem. I would call the things you identified on the previous message two separate problems.

commit 5724d8f5 (2017-12-20): posemath: Adapt upstream changes

This is based on inspection of rcslib-2014.04.29 but is not a direct
copy as the API was changed from taking pointers to taking copies
when it comes to inputs that are not changed.  Besides the issue
discussed in #379, one of the atan2 calls is changed.

Closes: #379