LinuxCNC/linuxcnc #384: sim/axis/vismach/puma does not home

rene-dev, 2017-12-24:
## Here are the steps I follow to reproduce the issue:

 1.  run sim/axis/vismach/puma
 2. Enable, Home all

## This is what I expected to happen:
Homing
## This is what happened instead:
kinematicsInverse failed
## It worked properly before this:
It worked at some point.

## Information about my hardware and software:
 * I am running A binary I built myself

possibly related to https://github.com/LinuxCNC/linuxcnc/commit/7609a0806e54acd394716dbb9d6b78d6e73abb5d

SebKuzminsky, 2017-12-25:
The problem was introduced in commit b336b8caa4, where I merged the puma fixes from 2.7 into master.  Both parents of the merge commit are good (2.7's 69d7a47b25 and master's af054c9cda), so the bug must have been introduced by an interaction of the puma fixes in 2.7 and the master branch that escaped my attention.

In master's af054c9cda (the last good commit in master, just before the buggy merge), the puma.ini vismach model start out pointing straight up, and moves to home.

In 69d7a47b25 (the commit in 2.7 where the puma fix was merged) the model starts out with the "upper arm" pointing to the side and the "fore arm" pointing down, and the arm does *not* move to home.

In b336b8caa4 (the commit in master that merged 2.7 and introduced the bug), the model starts out like in 2.7, does not home to move, and immediately dies from the "kinematicsInverse failed" error after switching to world mode.

andypugh, 2018-03-09:
Email report from Rudy duPreez:
--8X-----------------------------------------
OK, so i did a complete rebuild with some print statements in pumakins. It normally first calls the forwardkins a number of times after homing, and then when you jog an axis, it also calls invkins.

In 2.7 it does that and works normally, but in 2.8 it only calls forwardkins and I think at the first call to inversekins it stops with the error " inversekinematics failed". It does not enter inversekins, so there is some external problem to call inversekins in 2.8. I am afraid the problem is beyond my knowledge. It does not seem to be related to something in pumakins or posemath.
--8X-------------------------------------------

I suspect that this is related to Joints-Axes and which kins functions are called depending on KINEMATICS_BOTH and KINEMATICS_IDENTITY

closed 2018-11-10 by rene-dev