LinuxCNC/linuxcnc #387: feature request: track peak following error

SebKuzminsky, 2017-12-25:
It would be helpful and interesting to track the maximum following error.  Each joint should grow a new output float pin named something like ".ferror-max", which holds the highest ferror value seen on that joint since LinuxCNC was started.

fixerslo, 2018-07-02:
There should also be a way to reset the value. Maybe with machine-off -> machine-on transition. Or a .reset-ferror-max pin.

rene-dev, 2018-07-02:
why not do that in hal?

pcw-mesa, 2018-07-02:
If you mean track the largest FE in hal, its certainly possible but this is of such general utility
I think it would be better if it were native to motion, because its a great diagnostic tool for random 
errors on systems that may not have the appropriate hal wiring installed