LinuxCNC/linuxcnc #392: libnml should ask pkg-config about libtirpc location

l29ah, 2018-01-03:
As the internal glibc sunrpc implementation is deprecated long time ago and removed in the recent versions.
```
In file included from libnml/cms/cms.cc:37:0:
libnml/cms/cms_xup.hh:19:39: fatal error: rpc/rpc.h: No such file or directory
```
```
∞ pkg-config --cflags libtirpc       
-I/usr/include/tirpc
```

Petross404, 2018-02-02:
> ∞ pkg-config --cflags libtirpc       
-I/usr/include/tirpc

This works?

spike77453, 2018-05-02:
sunrpc seems to be removed from glibc >= 2.27 so linuxcnc doesn't compile anymore :(

jepler, 2018-05-03:
It appears that for master branch, it would be acceptable to change unconditionally to libtirpc, because the package libtirpc-dev is in ubuntu 12.04 and newer and debian 7.0 and newer.  However, a fix targeted to 2.7 must adapt to either situation, because 2.7 still targets ubuntu 10.04 which does not live package libtirpc-dev.

jepler, 2018-05-03:
.. I upated http://wiki.linuxcnc.org/cgi-bin/wiki.pl?MinimumSoftwareVersions with information about libtirpc-dev based on a bit of googling.  Feel free to update it (or just make a correction here) if I got it wrong.

Petross404, 2018-05-05:
> sunrpc seems to be removed from glibc >= 2.27 so linuxcnc doesn't compile anymore :(
I think it got removed one or two versions earlier ;)

awigen, 2019-05-11:
It was removed in 2011 glibc-2.14: https://sourceware.org/git/?p=glibc.git;a=commit;h=7b57bfe5988e476ea40934457dfd1c8a231e2391

rene-dev, 2019-05-13:
fixed by #587

closed 2019-05-13 by rene-dev