LinuxCNC/linuxcnc #400: M6 Remap does not handle pockets correct!

gmoccapy, 2018-01-20:
If you do remap M6 code, according to the stdglue.py file it should transfer as parameters 

self.params["tool_in_spindle"]
self.params["selected_tool"]
self.params["current_pocket"]
self.params["selected_pocket"]

But LinuxCNC does not handle the pocket relation correct.
Just place tool 3 in pocket 8 and call T3 M6, that should return pocket 8 but does return pocket 3 instead!

Any hint would be appreciated to solve this.

Norbert

andypugh, 2018-01-20:
On 20 January 2018 at 21:18, Norbert Schechner <notifications@github.com>
wrote:

> But LinuxCNC does not handle the pocket relation correct.
>
> Just place tool 3 in pocket 8 and call T3 M6, that should return pocket 8
> but does return pocket 3 instead!
>
It might work better if you turn on random tool changer.

The stdglue.py does say that the result is probably nonsense:
https://github.com/LinuxCNC/linuxcnc/blob/master/nc_files/remap_lib/python-stdglue/stdglue.py#L163

 The problem is that the tool table version passed about in NML does not
contain pocket info, and what you actually get (on a non-random changer) is
the index of the tool array.

It was one of the things I tried to fix in
https://github.com/LinuxCNC/linuxcnc/tree/andypugh/tooltable but I was
never brave enough to merge that without somebody else testing it, and
nobody ever did.

-- 
atp

gmoccapy, 2018-01-23:
I played a little bit with the code and pushed my changes as:

newbynobi__master_tool_pockets

I reached a state, that the pockets are reported correct at the remap module, but now the iocontrol pin are wrong :-(

As mentioned I am completely new to C++ so I may produce rubbish, but I am trying hard and Rene_dev told me I must learn C++, so here I am learning.

So could someone push me in the right direction?

Norbert

P.S. I added two configs (axis and gmoccapy) both containing a diagnostics and a rack tool changer tab. (axis is not so beautiful to embed the rack changer but does work after extending the screen.

To use the python tool changer just click on the open button to open the changer cover, after that you can click on an image to load or unload a tool (like you would do on a real machine)
Watch the tool file to see the pockets changing by 100 places.

rene-dev, 2018-01-23:
The code he is talking about is here: https://github.com/LinuxCNC/linuxcnc/compare/newbynobi_master_tool_pockets#diff-22efa6c95bb01316fb5517cec373f87dL25
hmm, iocontrol takes the numbers from  emcstatus https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/iotask/ioControl.cc#L876

gmoccapy, 2018-02-18:
Hallo Andy,

I just tried your branch, nut that one is based on 2.6.
I tried to merge master into that one, but than my configs does not start, may be it is related to the new Joint Axis handling :-(

Norbert

gmoccapy, 2018-02-22:
OK, after sebs last commit, iocontrol does report the correct pocket and toolno, but remap still shows very strange numbers.

Please open /sim/axis/remap/rack-toolchange/racktoolchange
Open tooltable and place T3 in pocket 5, T5 in pocket 6 and T6 in pocket4
Apply changes
Open hal show and watch iocontrol pin
Open /nc_subroutines/rack_toolchange.ngc and uncomment the print statements
Save file
Get out of estop / home machine, goto MDI
enter T1
enter M6
Tool will be picked from pocket 1, as it should, reporting in terminal is OK
enter T6
enter M6
Tool will be picket from pocket 6 instedt of pocket 4

Where is that handled?

Norbert

rene-dev, 2018-02-22:
I think @SebKuzminsky would really like a test for this.
what you see is most likely the index number. this can be converted by looking it up in the fms array.
in iocontrol this is done here with fms[p]:
https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/iotask/ioControl.cc#L869
but I dont know where the remap is called...

gmoccapy, 2018-02-22:
And we need also a working param[current_pocket], that one is also not handled correctly. Otherwise the remap would not know where to put back the tool in spindle.

But unfortunately I do not know easer where to look for that :-(

Norbert

gmoccapy, 2018-02-22:
See also issue 239, it is related to this one

rene-dev, 2018-02-25:
@gmoccapy if you write the numbers like this #239 the issues are referenced.

I think part of the problem is, that pocket numbers for non random toolchangers where never intended to be used like this:

https://github.com/LinuxCNC/linuxcnc/blob/15a4d11ffe8e190c7ba088be2de6650e9f0b0ccc/docs/src/code/code-notes.txt#L1442-L1444

rene-dev, 2020-04-20:
fixed in 2.8 and master by dab2a051b30e600ef95fbe911053386ef44631d0

closed 2020-04-20 by rene-dev