LinuxCNC/linuxcnc #411: Tool amount limit

gmoccapy, 2018-02-22:
With standard config linuxcnc only allows handling 55 tools. Even if you have more in the tool table, you will not be able to use more tools. this is related to the limited shared memory as far as I found.

To test use this tooltable and try T70 M6
~~~~~
T1 P1 D4.500000 Z-2.667780 ;4.5 mm 3 flute cutter
T2 P5 D25.000000 Z-17.295477 ;25 mm cutter
T3 P4 D7.500000 Z+1.229786 ;7.5 mm 3 flute cutter
T4 P3 D4.000000 Z-10.256000 ;4 mm Drill
T5 P2 D5.000000 Z-10.254000 ;5 mm Drill
T6 P10 D6.000000 Z-11.120000 ;6 mm Carbide burr
T7 P9 D6.500000 Z-14.589000 ;6.5 mm HSS burr
T8 P8 D12.000000 Z-14.458000 ;Cutter with 2 faces
T9 P6 D16.000000 Z-16.527474 ;Cutter with 3 faces
T10 P7 D165.000000 Z-12.112000 ;Facer with carbide inserts
T11 P11 D10.000000 Z-14.258000 ;Centering drill
T12 P12 D4.000000 Z+13.214000 ;3d taster elektronisch
T13 P13 D13.000000 Z-14.388000 ;3-Schneider Schaftfräser
T14 P14 D10.000000 Z-12.256000 ;3 Schneiden Stufenbohrer 5.5 auf 10
T15 P15 D10.000000 Z-14.888000 ;Bohrer TiN beschichtet
T16 P16 D12.500000 Z-14.583000 ;Stufenbohrer 6.8 auf 12.5 - 3 Schneiden
T17 P17 D6.500000 Z-14.254000 ;Stufenbohrer 4.2 auf 6.5 mit 45 Grad Übergang
T18 P18 D4.000000 Z-16.595000 ;Schaftfräser 4 Schneider
T19 P19 D11.000000 Z-15.235000 ;Bohrer
T20 P20 D50.100000 Z-16.235000 ;Messerkopf 4 Platten
T21 P21 D7.800000 Z-14.587000 ;Bohrer poliert
T22 P22 D16.150000 Z-18.253000 ;Messerkopf Zweischneider
T23 P23 D3.300000 Z-14.254000 ;Bohrer poliert
T24 P24 D12.020000 Z-14.256000 ;Fräser nachgeschliffen, nicht beschichtet
T25 P25 D4.500000 Z-2.667780 ;4.5 mm 3 flute cutter
T26 P26 D25.000000 Z-17.295477 ;25 mm cutter
T27 P27 D7.500000 Z+1.229786 ;7.5 mm 3 flute cutter
T28 P28 D4.000000 Z-10.256000 ;4 mm Drill
T29 P29 D5.000000 Z-10.254000 ;5 mm Drill
T30 P30 D6.000000 Z-11.120000 ;6 mm Carbide burr
T31 P31 D6.500000 Z-14.589000 ;6.5 mm HSS burr
T32 P32 D12.000000 Z-14.458000 ;Cutter with 2 faces
T33 P33 D16.000000 Z-16.527474 ;Cutter with 3 faces
T34 P34 D165.000000 Z-12.112000 ;Facer with carbide inserts
T35 P35 D10.000000 Z-14.258000 ;Centering drill
T36 P36 D4.000000 Z+13.214000 ;3d taster elektronisch
T37 P37 D13.000000 Z-14.388000 ;3-Schneider Schaftfräser
T38 P53 D10.000000 Z-12.256000 ;3 Schneiden Stufenbohrer 5.5 auf 10
T39 P54 D10.000000 Z-14.888000 ;Bohrer TiN beschichtet
T40 P55 D12.500000 Z-14.583000 ;Stufenbohrer 6.8 auf 12.5 - 3 Schneiden
T41 P56 D6.500000 Z-14.254000 ;Stufenbohrer 4.2 auf 6.5 mit 45 Grad Übergang
T42 P57 D4.000000 Z-16.595000 ;Schaftfräser 4 Schneider
T43 P58 D11.000000 Z-15.235000 ;Bohrer
T44 P99 D50.100000 Z-16.235000 ;Messerkopf 4 Platten
T45 P59 D7.800000 Z-14.587000 ;Bohrer poliert
T46 P60 D16.150000 Z-18.253000 ;Messerkopf Zweischneider
T47 P61 D3.300000 Z-14.254000 ;Bohrer poliert
T48 P62 D12.020000 Z-14.256000 ;Fräser nachgeschliffen, nicht beschichtet
T49 P63 D4.500000 Z-2.667780 ;4.5 mm 3 flute cutter
T50 P64 D25.000000 Z-17.295477 ;25 mm cutter
T51 P65 D7.500000 Z+1.229786 ;7.5 mm 3 flute cutter
T52 P66 D4.000000 Z-10.256000 ;4 mm Drill
T53 P67 D5.000000 Z-10.254000 ;5 mm Drill
T54 P68 D6.000000 Z-11.120000 ;6 mm Carbide burr
T55 P69 D6.500000 Z-14.589000 ;6.5 mm HSS burr
T56 P70 D12.000000 Z-14.458000 ;Cutter with 2 faces
T57 P71 D16.000000 Z-16.527474 ;Cutter with 3 faces
T58 P72 D165.000000 Z-12.112000 ;Facer with carbide inserts
T59 P11 D10.000000 Z-14.258000 ;Centering drill
T60 P12 D4.000000 Z+13.214000 ;3d taster elektronisch
T61 P13 D13.000000 Z-14.388000 ;3-Schneider Schaftfräser
T62 P14 D10.000000 Z-12.256000 ;3 Schneiden Stufenbohrer 5.5 auf 10
T63 P15 D10.000000 Z-14.888000 ;Bohrer TiN beschichtet
T64 P16 D12.500000 Z-14.583000 ;Stufenbohrer 6.8 auf 12.5 - 3 Schneiden
T65 P17 D6.500000 Z-14.254000 ;Stufenbohrer 4.2 auf 6.5 mit 45 Grad Übergang
T66 P18 D4.000000 Z-16.595000 ;Schaftfräser 4 Schneider
T67 P19 D11.000000 Z-15.235000 ;Bohrer
T68 P20 D50.100000 Z-16.235000 ;Messerkopf 4 Platten
T69 P21 D7.800000 Z-14.587000 ;Bohrer poliert
T70 P22 D16.150000 Z-18.253000 ;Messerkopf Zweischneider
T71 P23 D3.300000 Z-14.254000 ;Bohrer poliert
T72 P84 D12.020000 Z-14.256000 ;Fräser nachgeschliffen, nicht beschichtet
~~~~~
to increase the tool amount you need to edit two files, i.e. 256 tools:

/configs/common/linuxcnc.nml
change:
emcStatus from 16384 to 33792
toolSts from 8192 to 33792

and /src/emc/nml_intf/emctool.h
CANON_POCKETS_MAX to 257

IMHO we should follow rene-dev and change the tool handling to use a sqlite database. This way we have also the possibility to add more information, like tool wear, tool use time, cutting speed, feed, etc..

Norbert

aShure, 2018-05-10:
Thanks for the workaround!
I agree, a new tool handling with statistics, feed per tooth,.. would be awesome.

FrancescoTalotta, 2018-07-05:
Hello,

I tried your workaround but I get a lot of errors, mainly regarding this:

`libnml/buffer/physmem.cc 143: PHYSMEM_HANDLE: Can't write 33728 bytes at offset 60 from buffer of size 33760.
libnml/cms/cms_in.cc 1378: CMS:(emcStatus) Error writing 33728 bytes to global memory at offset 0x8e39fd8`

rene-dev, 2020-04-07:
mitigated by https://github.com/LinuxCNC/linuxcnc/commit/b51ef8cc3c560b6c44d095814988a3f972bc0763

closed 2020-04-07 by rene-dev