LinuxCNC/linuxcnc #419: When using Pumakins in Master Kinematics_Inverse is never called

andypugh, 2018-03-09:
Email report from Rudy duPreez:

OK, so i did a complete rebuild with some print statements in pumakins. It normally first calls the forwardkins a number of times after homing, and then when you jog an axis, it also calls invkins.

In 2.7 it does that and works normally, but in 2.8 it only calls forwardkins and I think at the first call to inversekins it stops with the error " inversekinematics failed". It does not enter inversekins, so there is some external problem to call inversekins in 2.8.  I am afraid the problem is beyond my knowledge. It does not seem to be related to something in pumakins or posemath.

rene-dev, 2018-03-09:
maybe related to #384 or #390?

andypugh, 2018-03-09:
Merge with #384

closed 2018-03-09 by andypugh