LinuxCNC/linuxcnc #422: HAL gladevcp Iconview widget - Coding error

fairnorth, 2018-03-19:
Small things can bug me a little because you never know if these small things will lead to other more serious issues later. I decided to look at why my gmoccapy always reports an error at start up, Invalid icon size 48. Not a big issue but still a problem that needs to be fixed. One of the causes for this is that Hal GladeVCP iconview.py has a programming error and thus more than just gmoccapy reports can show this error in its printed log.

Here is my description of the issue with Hal GladeVCP iconview.py and a simple suggestion for a fix.

Withing the iconview.py are 9 lines like:

"image.set_from_stock(gtk.STOCK_GOTO_TOP, 48)"

The 48 is supposed to refer to the displayed file folders icons desired size. This cannot be done with entering 48 in these lines.
In these lines size is looking for a "name" that has defined an icon and "not a size number".

Per the gtk Class Reference for gtk.Image, the normal allowed things for unless a custom one is made are:

gtk.ICON_SIZE_MENU
gtk.ICON_SIZE_SMALL_TOOLBAR
gtk.ICON_SIZE_LARGE_TOOLBAR
gtk.ICON_SIZE_BUTTON
gtk.ICON_SIZE_DND
gtk.ICON_SIZE_DIALOG.

Here are my suggested changes to consider to the Hal GladeVCP iconview.py widget:

Leave this line alone:
self.icon_size = 48 # The actual per file folder display size - must agree with the named item standard size definitions. A error in choosing the wrong size here will show wrong folder colors since that size is not likely defined as being possible in the named STOCK item definition.

Ok now for the suggested changes:

Add this line:
stksize = gtk.ICON_SIZE_MENU # GTK standard size name # chosen at random

Change 9 lines from and to list:
image.set_from_stock(gtk.STOCK_HOME,48)
image.set_from_stock(gtk.STOCK_HOME, stksize)

image.set_from_stock(gtk.STOCK_GOTO_TOP,48)
image.set_from_stock(gtk.STOCK_GOTO_TOP,stksize)

mage.set_from_stock(gtk.STOCK_GO_BACK,48)
image.set_from_stock(gtk.STOCK_GO_BACK, stksize)

image.set_from_stock(gtk.STOCK_GO_FORWARD,48)
image.set_from_stock(gtk.STOCK_GO_FORWARD, stksize)

#image.set_from_stock(gtk.STOCK_GO_DOWN,48)
#image.set_from_stock(gtk.STOCK_GO_DOWN, stksize)

#image.set_from_stock(gtk.STOCK_GO_UP,48)
#image.set_from_stock(gtk.STOCK_GO_UP, stksize)

image.set_from_stock(gtk.STOCK_JUMP_TO,48)
image.set_from_stock(gtk.STOCK_JUMP_TO, stksize)

image.set_from_stock(gtk.STOCK_OK,48)
image.set_from_stock(gtk.STOCK_OK, stksize)

image.set_from_stock(gtk.STOCK_STOP,48)
image.set_from_stock(gtk.STOCK_STOP, stksize)

This eliminates the Invalid icon size 48 error message generated by this widget.
I have seen similar errors in the past from at least one other file but cant recall its name.

andypugh, 2018-03-19:
Is this one for you, Chris?

KurtJacobson, 2018-03-19:
@fairnorth I remember getting `invalid icon size` warnings when running Gmoccapy on wheezy, but I have not noticed them when running Gmoccapy 1.5.9 as shipped with LCNC 2.7.12. What OS and versions of LCNC and Gmoccapy are you running?

It has been a long time since I used the IconView file chooser, but IIRC you can set the icon size from within the glade editor by changing an INT value, which I assume is held in `self.icon_size`. Do the changes you suggested have any effect on that ability?

I am pretty sure @gmoccapy is the author of iconview.py, and since Gmoccapy is the main place the icon chooser is used it might be best for him to look into this.

KurtJacobson, 2018-03-19:
Ah, now I see the GtkWarnings, I was not looking close enough! I thought GtkWarnings showed up as bold and yellow in the terminal, but I guess that is only with Gtk3. I am spoiled rotten :grin:

c-morley, 2018-03-19:
Fix pushed to master - If there is still a problem not covered please reopen. Thanks everyone.

closed 2018-03-19 by c-morley