LinuxCNC/linuxcnc #424: rs274: erroneously (?) accepts #abc>=7

jepler, 2018-03-19:
Named variables are documented as having the form `#<abc>`, but in some contexts, the leading `<` is not actually checked.  Specifically, the following is accepted but probably shouldn't be:
~~~~
$ ./bin/rs274 -g
executing
    1 N..... USE_LENGTH_UNITS(CANON_UNITS_MM)
    2 N..... SET_G5X_OFFSET(1, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000)
    3 N..... SET_G92_OFFSET(0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000)
    4 N..... SET_XY_ROTATION(0.0000)
    5 N..... SET_FEED_REFERENCE(CANON_XYZ)
READ => #abc>=7
READ => 
~~~~

@cradek stated on IRC:
~~~~
14:36:23  <cradek> if((line[*counter] == '<') || isalpha(line[*(counter)]))
14:36:57  <cradek> I think this is just a bug
14:37:16  <cradek> a mistake from before he decided on requiring <...>
~~~~

It does not seem possible to *REFER* to such a variable (`Z#abc`, `Z[#abc>]`, `Z#<abc>` and other possibilities  that might refer to the created variable don't work.) This means the impact of fixing it is low: it would only affect the assignment of a variable that could never be referred to.  It also brings the implementation in line with the documentation.

cradek, 2018-03-20:
Then a little later I figured out more:

```
14:39 <cradek> the character after # can be any isalpha and it still works as if that character is <
14:39 <cradek> #Qabc>=1
14:39 <cradek> #<abc>=1
14:39 <cradek> these are equivalent
14:40 <cradek> so #O<sub>=3 defined the named param "<sub" to be 3 (yes < are allowed in param names)
```

So it is possible to refer to the variable in your example as `#<bc>` and maybe other ways too.  I agree it's unlikely fixing this will break any programs.  I also suspect that `<` shouldn't be allowed as part of a variable's name.

SebKuzminsky, 2018-03-20:
Good find!
The [docs](http://linuxcnc.org/docs/2.7/html/gcode/overview.html#_named_parameters) say:
~~~~~
Named parameters must be enclosed with < > marks.
~~~~~
 So I agree that this is a bug in the interpreter.  I think we should fix it in 2.7 even though it might break existing g-code programs, because any such g-code programs are wrong (ie are in violation of the syntax of our g-code language).

jepler, 2018-03-20:
Fixed by #425.