LinuxCNC/linuxcnc #426: Arc Blending Bug

ToddZuercher, 2018-03-21:
The New Trajectory Planner pauses under certain circumstances.
https://forum.linuxcnc.org/20-g-code/34213-axis-short-stop-very-short-about-every-15-20-sec-when-cutting-circle?start=10#107672

## Here are the steps I follow to reproduce the issue:

 1. A helical arc, followed by 3 or more regular arcs.
 
## This is what I expected to happen:
Smooth blended motion between all the moves.

## This is what happened instead:
Linuxcnc will pause between the 1st and 2nd regular arcs. (Decelerate to 0 then continue.)

## It worked properly before this:
? Disabling the new TP with the ini setting ARC_BLEND_ENABLE = 0 makes the pause go away.

## Information about my hardware and software:
Running on multiple systems (real hardwar and VMs) using the latest Wheezy, and Stretch Linuxcnc images with the latest 2.7 version of Linuxcnc (from Buildbot.)

pcw-mesa, 2018-03-21:
I see the same pauses on that gcode with the new TP disabled, on both 2.7 and master
so if there's an issue, it seems to be common to both TPs

ToddZuercher, 2018-03-21:
Did you try running the simplified G-code I posted with both planners?
https://forum.linuxcnc.org/media/kunena/attachments/3190/arc-bug.ngc
Try it with in G20 with about F150 (or how ever fast you can on the config your testing with where the acceleration constraints won't get in the way too much.) 
The difference between the new and old TP is startling.

pcw-mesa, 2018-03-21:
Yes, and I get the same results as I had before, that is both the new and old TPs have similar
glitches
Though in some cases they dont occur on old vs new TP and this is very feed rate dependent

robEllenberg, 2018-03-21:
I can confirm this is a TP bug, luckily one I have a fix for in a local branch. It's a regression in one of the later 2.7 releases. What happens is that the TP fails to create an arc blend between segments (because they aren't planar), but fails to fall back to parabolic blending. Instead, it forces the blend to be treated as tangent, which means it needs to slow way down at the corner to obey acceleration constraints. 

I'll send word as soon as I have a hotfix branch ready (hopefully this weekend, the next few days are quite busy).

andypugh, 2018-10-09:
Is this ready for merge?

rene-dev, 2018-10-17:
https://github.com/LinuxCNC/linuxcnc/pull/509

rene-dev, 2018-11-06:
This should be fixed in 2.7 and 2.8-pre now.

closed 2018-11-06 by rene-dev