LinuxCNC/linuxcnc #428: Inconsistent G33.1 behavior between implementation and the doc.

ReeseWang, 2018-04-11:
## Here are the steps I follow to reproduce the issue:

 1. Move to G54 X0 Y0 Z0.
 2. "M3 S300" to start the spindle.
 3. Execute "G33.1 X10 Y0 Z-10 K0.7"

## This is what I expected to happen:
The mill rapid moves to X10 Y0 Z0 and then does a rigid tap which has a depth to Z-10.
Just as the Warning in G33.1 documentation says:

> If the X Y coordinates specified are not the current coordinates when calling G33.1 for tapping the move will not be along the Z axis but will rapid move from the current location to the X Y location specified.

## This is what happened instead:
The mill did a linear spindle-synchronized move from X0 Y0 Z0 to X10 Y0 Z-10, which broke my tap.
I'm sure the rigid tapping is working normally if I do "G0 X10 Y0" first and then "G33.1 Z-10 K0.7".

## It worked properly before this:
(N/A)

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): Not relevant I think.
 * I am using this kernel version (shown by `uname -a`): Not relevant I think.
 * I am running ...
   * [x] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): 2.7.12
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): AXIS
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): Parallel Port

SebKuzminsky, 2018-04-11:
Yikes!

I think the expected motion of your `G33.1 X10 Y0 Z-10 K0.7` should be equivalent to:
~~~~~
G0 X10 Y0
G33.1 Z-10 K0.7
~~~~~

This motion would be analogous to (for example) `G81 X10 Y0 Z-10`.

I'll look into it.

jethornton, 2018-04-11:
Docs should say it will tap on a traverse move from start to specified location. AFAIK it's been like that from the start.

SebKuzminsky, 2018-04-11:
I, too, think that the observed behavior has been around for a long time.

Updating the docs to match this observed behavior is probably a good start, but is that behavior what we want it to be?

On a mill i think "diagonal tapping" is not ever useful, but maybe on a lathe it is, for single-point thread-cutting a tapered tread?

SebKuzminsky, 2018-04-11:
@cradek educated me some about this in IRC: G33.1 is not meant to be limited to tapping along Z, it is intentionally able to tap in any direction.  This is not useful on an XYZ mill, but it *is* useful on 4- and 5-axis mills.  It's the documentation/warning that is wrong.

SebKuzminsky, 2018-04-11:
Code inspection reveals that G33.1 (rigid tap) only accepts XYZ, not ABC or UVW axis words.  So it's not the case that the current behavior can be used on 5-axis mills.  G33 (without the .1) does accept all 9 axis words, and might be used on 4+ axis mills for spindle-synchronized motion along arbitrary vectors.

jethornton, 2018-04-11:
Documents fixed.

closed 2018-04-11 by jethornton