LinuxCNC/linuxcnc #429: Hm2_rpspi and hm2_rpi drivers fail

JR1050, 2018-04-17:
## Here are the steps I follow to reproduce the issue:
Using the rpspi driver with the latest 2.8 pre produces the following error. Pi3 is connected to 7i90 board via spi.  Lcnc invoked from the terminal

hm2_rpspi: SPI0/CE0 Invalid cookie, read: 2ad5e5ff 2a292724 192a2726, expected: 55aacafe 54534f48 32544f4d
hm2_rpspi: SPI0/CE0 MISO input is bit-shifted by one bit. SPI read clock frequency probably too high.
hm2_rpspi: rtapi_app_main: No such device (-19)
./hnc112517.hal:24: waitpid failed /usr/bin/rtapi_app hm2_rpspi
./hnc112517.hal:24: /usr/bin/rtapi_app exited without becoming ready
./hnc112517.hal:24: insmod for hm2_rpspi failed, returned -1
Shutting down and cleaning up LinuxCNC...
hm2_rpspi: not loaded
<commandline>:0: exit value: 255
<commandline>:0: rmmod failed, returned -1
hm2: unloading
<commandline>:0: unloadrt failed

 1. I tried changing the spi clock rates, no effect
 2. Tried the hm2_rspi driver and got different errors
 3. Tried my old rpspi driver from last year, same errors as the rspi driver

## This is what I expected to happen: it to work 

## This is what happened instead: see above error

## It worked properly before this:
Using an older about a year old version of 2.8 and an older rpspi driver
It worked and was running a machine

## Information about my hardware and software:
Rpi3 running raspian, rtprempt patch, latest 2.8 pre. Mesa 7i90 board with encoder/ pwm/ ssi bit file svt_8 I think . Hm2_rpspi was compiled in place with latest source file and header files from GitHub.  2.8 was pulled from buildbot

 
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): Axis
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card):  mesa 7i90 with spi

andypugh, 2018-04-18:
Are you using the very latest firmware from Mesa? There was a bug in the LinuxCNC driver that masked a bug in the 7i90 firmware. 
We fixed the LinuxCNC bug (affected all cards Smart-Serial cards) but this necessitated a re-release of the Mesa firmware.

pcw-mesa, 2018-04-18:
I don't think this is firmware related since the firmware was working before a LinuxCNC update, and the error is quite low level (my guess is either a clock speed or clock polarity change in the driver) The firmware bug that triggered issue 412 (uncovered by the sserial bug fix ) only affects when SPI writes are done

andypugh, 2018-04-24:
Where did you get your Pi kernel and LinuxCNC version from? I have a Pi and 7i90 so ought to be able to reproduce your hardware.

JR1050, 2018-04-25:
Andy, thank you for looking into this. My kernel came from here
http://www.frank-durr.de/?p=203, I had tried to do it from scratch , but I had trouble, it was easier to use his.  My lcnc came from buildbot, the rpspi driver and header files from git.

rmu75, 2018-05-22:
just FYI, kernel 4.9.65 with rt preempt patch 56v7 works reasonably well.

Can you show the parameters to to hm2_rpspi? I find that read-frequencies above 30MHz won't work, this is with ca. 50mm SPI connection and serial termination.

Actual line from working configuration: `loadrt hm2_rpspi spiclk_rate=60000 spiclk_rate_rd=29000` on a somewhat recent linuxcnc 2.8 built on the Pi.

pcw-mesa, 2018-07-14:
I can verify that 4.9.65 works (thanks rmu for the kernel debs!) with hm2_rpspi and LinuxCNC 2.8
(2.8 built a couple days ago) This is not with a 7I90 but a 7C80 which has a RPI compatible GPIO connector, but the SPI firmware interface is identical

rmu75, 2018-11-12:
This weekend i upgraded to 

* kernel 4.14.74-rt44-v7 built from https://github.com/raspberrypi/linux/tree/rpi-4.14.y-rt
* linuxcnc master as of 2018-11-10
* latest mesa BIT File (7i90_spi_svss6_8.bit)
* raspberry pi 3b+

Seems to work flawlessly.

I think this bug can be closed if there is no further input.

closed 2018-11-12 by rene-dev