LinuxCNC/linuxcnc #430: XZ Machine, Limit Switch Error

MarcelineVQ, 2018-04-20:
When creating a default XZ machine in Stepconf and selecting "All limits + homes" option for your limit switch pin the hal settings for homing will look like this:
`net homing-x <= axis.0.homing => lut5.0.in-0`
`net homing-z <= axis.1.homing => lut5.0.in-1`

This causes a limit switch error in axis from an incorrect lut5 result because the second homing value should actually be
`net homing-z <= axis.2.homing => lut5.0.in-1`
Since in the ini file and even later in the hal file the second joint of an XZ setup is 2, not 1.

It seems to me that
https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/usr_intf/stepconf/build_HAL.py#L262
is responsible for assigning the 1 instead of 2 but I've not a dev machine to confirm with.

c-morley, 2018-04-22:
Ok I pushed a fix to 2.7 - pncconf was affected too - thank you for the report.
I would fix master too but there is a merge conflict with hostmot2 that I don't know the right course of action. Once that is fixed I will fix master.

c-morley, 2018-04-22:
I might add if you can test the fix and report back that would be great.

c-morley, 2018-04-22:
ok I merged to master..but actually master should not be affected because of using joints.
Joints don't need to skip numbers like axes do plus the code around there was rewritten.
There may still be a problem with stepconf with foam configs ( I see stuff like joints 6 and 7 which probably should be 3 and 4) but i'll have to look at it later.

SebKuzminsky, 2018-04-22:
Oops, I think a problem crept in during the hm2_7i90 merge.  I'll fix it.

MarcelineVQ, 2018-04-24:
Thank you! This does fix the homing issue, but you'll need to change
`loadrt stepgen step_type=0,0` to `loadrt stepgen step_type=0,0,0` if you meant to change the stepgen to 2 instead of 1 later in the hal file.
And thanks for your prompt attention on this issue by the way, I've got a dev machine set up now so hopefully I can just make PR's for issues that come up down the road instead of making you guys do it hehe.

rene-dev, 2018-11-10:
can this be closed?

closed 2018-11-10 by MarcelineVQ