LinuxCNC/linuxcnc #435: reference to nonexistent files in CONTRIBUTING

rene-dev, 2018-05-04:
docs/SubmittingPatches and docs/developer-certificate-of-origin are referenced in https://github.com/LinuxCNC/linuxcnc/blob/master/CONTRIBUTING.md
but they dont exist.

rene-dev, 2018-05-04:
style guide should be linked somwehere: http://linuxcnc.org/docs/2.7/html/code/style-guide.html

rene-dev, 2018-05-04:
contributing seems to be here: http://linuxcnc.org/docs/2.7/html/code/contributing-to-linuxcnc.html

jepler, 2018-05-04:
I'm closing this issue because the item mentioned in the issue's title is resolved by 7a1bfbcfcf9d31d92e52d1dc8674840f54ea54b4 -- this does not mean that the other issues mentioned in the comments are unimportant, but as an issue has only one open/closed state I prefer to close this one.  Please create further issues or pull requests for these other problems you have identified.

closed 2018-05-04 by jepler