LinuxCNC/linuxcnc #436: Allow cycles like G81 in both positive and negative

aShure, 2018-05-10:
LinuxCNC's G81 allows operations into the working plane only, not out of the working plane. Example: A large back-counterbore is to be drilled. With this op the retract plane is further down in Z- than the actual machining point. LinuxCNC won't accept a retract plane lower than the machining point.

Another example: A tool in the spindle has got a 90 deg offset in A or B, for horizontal boring operations. G81 works fine if approached from X+, drilling into the right side of the workpiece, but breaks if approached from X-, left side of the work piece. Example code:

```
(DRILL FROM LEFT)
(TOOL AT B-90 ANGLE WITH ANGLE ADAPTER, NOT WITH ROTARY TABLE, HORIZONTAL OP FROM LEFT SIDE )
(SET G19, WORKPLANE YZ, APPROACH FROM LEFT SIDE)
N100 G19 (CURRENTWORKPLANE 0, -1.5708, 0)
N105 G0 X-5.
N110 G81 X10. Y-20. Z-20. R5. F250
N115 G0 X-15.
N120 G53 Z0.
```

aShure, 2018-05-10:
to be labeled as an enhancement by @andypugh  :)

SebKuzminsky, 2018-06-10:
Are these operations for "drilling" with a boring bar in a lathe?

andypugh, 2018-06-10:
Back-counterboring with a boring head in a mill would be one application. 
I can imagine you would need it with a rotating head milling machine and wanted to drill from both sides of the material too (drilling in X in the YZ plane)

aShure, 2018-06-11:
Example: VMC with a four directional horizontal spindle attachment
[image]