LinuxCNC/linuxcnc #440: motion.requested-vel unexpected behaviour

pcw-mesa, 2018-06-08:
# The issue tracker is not a support forum

The LinuxCNC issue tracker is to report bugs in the software.
If you have a question about how to use the software, use one of the other methods detailed on our community support page: http://linuxcnc.org/community/

(delete this section before submitting your bug report)

## Here are the steps I follow to reproduce the issue:

 1.
 2.
 3.

## This is what I expected to happen:

## This is what happened instead:

## It worked properly before this:
(If the behavior changed after making a particular change in hardware or
software, describe the change you think is responsible.  E.g., "after upgrading
from LinuxCNC 2.7.3 to 2.7.4")

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):
 * I am using this kernel version (shown by `uname -a`):
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card):

pcw-mesa, 2018-06-08:
motion.requested-vel is used by at least a couple of plasma Torch Height Control (THC) systems to allow setting of the corner lock threshold as a percent of feed rate. Unfortunately, motion.requested-vel is the TPs requested velocity and not the gcodes specified feed rate 
so the THC velocity compare can fail in some circumstances. One solution would be to add a new pin to motion that just reported the feed rate from gcode

SebKuzminsky, 2018-06-08:
I think this feature will require the state tags branch to do correctly.

SebKuzminsky, 2018-06-08:
... Or maybe not.  Maybe `tpActivateSegment()` could just put `tc->reqvel` on a pin.  And when the `tcq` drains, set that pin to 0.

SebKuzminsky, 2018-06-11:
@rodw-au, neat!  Do you have a branch somewhere that I can look at?

rodw-au, 2018-06-11:
Seb, let me see if I can learn how to do that after I do some more testing to make sure it works. I need to redo it on master branch instead of dgarr/external_offsets as well.

SebKuzminsky, 2018-06-12:
This is all to say: Motion doesn't know the programmed feed rate, it only knows the requested velocity of each segment.  This requested velocity is generated by Canon (the interface between the Interpreter and Motion, inside Task) based on machine constraints.

So I'm back to my knee-jerk reaction of believing this has to wait for State Tags.

I'm not familiar with Torch Height Control, can someone educate me on this use case where the programmed feed rate is more useful than either of the two velocities (motion.current-vel and motion.requested-vel) that are currently available?

pcw-mesa, 2018-06-12:
I am no expert but what I believe is wanted is a way to compare the gcodes programmed feed rate (FXXXX) which is the desired plasma cutting speed with the actual speed (current velocity) so the THC can be disabled (frozen at current height) when the current speed dips to a specified percentage of the programmed feed rate (to avoid diving at corners).

rodw-au, 2018-06-12:
Torch Height COntrol (THC) is fundamental to plasma operations. With plasma there is a linear relationship between torch voltage and cut height. Manufacturer's publish cut charts stating ideal heights, cut speed and arc voltage for a given material and thickness. Feedrate is usually set based on these parameters. The arc voltage is used as feedback to a PID controlling Z axis height. When the TP slows current-vel, voltage increases so the PID lowers the torch height sometimes resulting in a crash. The solution is to inhibit PID control when current_vel  falls below say 85% of feed rate set from the manufacturer's data. (Velocity Hold)

This issue stems from incorrect documentation stating that requested-vel is the F parameter.
At the high speeds used in plasma, changes to requested-vel are often violent.

[image]

Where a user depends on the incorrect documentation, THC operation is enabled unexpectedly due to the fall in requested_vel for a segment and a violent and unexpected correction can occur resulting in a divot in the cut.

Because of the current current-vel behaviour, the solution is to send the feed rate from Gcode twice, once as a F code and again as a M67/M68. Such data duplication is highly undesirable as it can result in errors.

I have spent several hours studying the code and came to understand the relationships between the modules as Seb has described. I noted that EMCMOTSTATUS was described global in the source and I did try setting a new variable in the structure from within canon (from SET_FEED_RATE procedure) but got a runtime error so I assume it is only global to motion. 

Can global EMCMOTSTATUS data be altered from canon?
Is there a branch for State Tags? 
Can you explain what State Tags does?

pcw-mesa, 2018-06-12:
I'm not so sure that the documentation is incorrect, just that it is incomplete 
That is motion.requested-vel is whats requested of the TP, which means it has 
additional constraints compared to the bare gcode feedrate number

To quibble, it really should be named a speed, not a velocity...

rodw-au, 2018-06-13:
2.8 docs at http://linuxcnc.org/docs/devel/html/config/core-components.html#sec:motion-pins
state:
**motion.requested-vel** - (float, out) The current requested velocity in user units per second from the F=n setting in the G Code file. No feed overrides or any other adjustments are applied to this pin.

Speed and velocity aside, there is nothing said about it being adjusted by Machine and TP constraints on a per segment basis.

SebKuzminsky, 2018-06-13:
I updated the docs (manpage and html docs, in 2.7 and master) to try to be more clear.

@pcw-mesa, I agree it's a speed, not a velocity, but I think calling it `requested-speed` would be more confusing, since "speed" is a term-of-art for the rotational speed of the spindle.  We also can't call it `requested-feed`, since that would exclude rapids...

rodw-au, 2018-06-13:
Can you give a synopsis of what "State Tags" does? I found the discussion about merging it but still none the wiser what it will achieve.

For plasma use, I think its possible to synthesise a feed-rate in a component that monitors requested-vel and has an output pin which is ratcheted up whenever requested-vel increases and is reset to zero when motion stops.

Thanks for updating the docs. I can see the change now.

rodw-au, 2018-06-14:
I'm just a novice with all of this but I think I just found there is a simple workaround. Perhaps plasma users should just remap Fnn and issue an M68 E1 Pnn to send the raw feedrate to HAL. Are there any possible negative side affects of this approach?

andypugh, 2018-06-14:
I hadn't realised that F could be remapped, but it appears that it can: http://linuxcnc.org/docs/2.7/html/remap/remap.html#remap:remappable-codes

This should work well, as it does not need the postprocessor to be THC-aware.

samcoinc, 2018-06-14:
Huh - I didn't know F could be remapped either..  cool - I think I have an application for that too.

sam

rodw-au, 2018-06-16:
Just to confirm that remapping the F code is a solution thats been tested by a couple of us and it will work. But before we close the issue and while we've been updating the docs earlier in this issue, may I point out that the remapping documentation is poorly lacking and the F code sction has several TBD's in it which  I assume means TO BE DONE.

Here is an example gcode setfeed.ngc sub based on the axis/remap sim that works, maybe it could be incorporated into the docs:
`o<setfeed> sub
(debug, in setfeed feed=#<feed>)
; using the code being remapped here means 'use builtin behaviour'
M67 E1 Q[#<feed>/60]
F#<feed>
; signal success be returning a value > 0:
o<setfeed> endsub [1]
m2
`